CadQuery/cadquery #354: How to perform a Minkowski sum with CQ?

lalebarde, 2020-05-20:
Hi, unless I am wrong, there is no _Minkowski sum_ function in CQ and it is not in the [roadmap](https://pythonhosted.org/cadquery/roadmap.html). Is there somewhere some alpha code for it or a workaround please?

adam-urbanczyk, 2020-05-20:
Nope, Minkowski sum is not something you can do with a BREP kernel as far as I know. What are you trying to achieve?

lalebarde, 2020-05-21:
Something like this:

[image]

```
import numpy as np
import matplotlib.pyplot as plt
from shapely.geometry import Point, Polygon, LineString, LinearRing
from shapely.affinity import translate
from shapely.ops import nearest_points, unary_union
from descartes import PolygonPatch
#from figures import BLUE, GRAY, set_limits

r0 = LinearRing([(0, 0), (1000, 0), (1000, 300), (0, 300)])
r = translate(r0, xoff=200, yoff=100, zoff=0)
print(r.wkt)
s = LinearRing([(-50, -50), (50, -50), (50, 50), (-50, 50)])
print(s.wkt)
n = nearest_points(s, r)
print([p.wkt for p in n])
l = LineString(n)
u = unary_union([s, l, r]).buffer(100)

fig = plt.figure()
ax = fig.add_subplot(111)

ax.plot(*np.array(r).T, color='blue', linewidth=3, solid_capstyle='round')
ax.plot(*np.array(s).T, color='green', linewidth=3, solid_capstyle='round')
ax.plot(*np.array(l).T, color='red', linewidth=3, solid_capstyle='round')
patch = PolygonPatch(u, fc='0.9', ec='0.5', alpha=0.5, zorder=2)
ax.add_patch(patch)
ax.axis('equal')

plt.show()
```

adam-urbanczyk, 2020-08-14:
If you want to offset profiles and then extrude, you could use `offset2D`.

closed 2020-08-14 by adam-urbanczyk