CadQuery/cadquery #355: cutThruAll failure

adam-urbanczyk, 2020-05-21:
The following code fails:
```python
import cadquery as cq

sphere_r = 10.

result = (cq.Workplane("XY").sphere(sphere_r)
            .workplane(centerOption="ProjectedOrigin").circle(sphere_r / 2.).cutThruAll()
            .workplane(centerOption="ProjectedOrigin").transformed(rotate=(90, 0, 0))
            .circle(sphere_r / 2.).cutThruAll()
            .workplane(centerOption="ProjectedOrigin").transformed(rotate=(0, 90, 0))
            .circle(sphere_r / 2.).cutThruAll()
            .workplane(centerOption="ProjectedOrigin").circle(sphere_r / 2.).extrude(sphere_r + 2, both=True))
```

perturbaing one of the angles makes it work

```python
import cadquery as cq

sphere_r = 10.

result = (cq.Workplane("XY").sphere(sphere_r)
            .workplane(centerOption="ProjectedOrigin").circle(sphere_r / 2.).cutThruAll()
            .workplane(centerOption="ProjectedOrigin").transformed(rotate=(90, 0, 0))
            .circle(sphere_r / 2.).cutThruAll()
            .workplane(centerOption="ProjectedOrigin").transformed(rotate=(0, 89.9, 0))
            .circle(sphere_r / 2.).cutThruAll()
            .workplane(centerOption="ProjectedOrigin").circle(sphere_r / 2.).extrude(sphere_r + 2, both=True))

show_object(result)
```

RubenRubens, 2021-01-13:
None of this works now. Using an old release (CQ-editor compiled for Windows) the second code works as expected.

**Error code**
```
Traceback (most recent call last):
  File "/testing/src/cutThruAll/C2.py", line 7, in <module>
    result = (cq.Workplane("XY").sphere(sphere_r)
  File "/cadquery/cadquery/cq.py", line 3105, in cutThruAll
    faceRef = sel2.filter(faces)[0]
  File "/cadquery/cadquery/selectors.py", line 103, in filter
    return [min(objectList, key=dist)]
ValueError: min() arg is an empty sequence
```

_Tested with (almost) the latest release using Docker. [Find the docker image here](https://github.com/RubenRubens/cq-testing/packages/568871)._

adam-urbanczyk, 2021-01-14:
Thanks, will try to fix this ASAP.

PR #604 (2021-01-26): cutThruAll fix for non-planar faces
This will resolve #355
closed 2021-01-26 by adam-urbanczyk