CadQuery/cadquery #357: Allow to define workplane using normal vector

adam-urbanczyk, 2020-05-21:
It would be nice to have this for both `cq.Workplane()` constructor and `workplane()` method. Everything can be done using a `transformed` call but is sometimes unnatural.

nopria, 2020-08-06:
I came across your comment while examining the functionalities of Plane object as a coordinate system. Probably I don't understand well your comment, but you can define a Workplane using 3-tuples:

```
import cadquery as cq
wp = cq.Workplane(cq.Plane(origin=(0,0,0), xDir=(1,1,0), normal=(1,0,0)))

```
You can use this code to show the Workplane coordinate system:

```
my_format = lambda n:'{:<8}'.format(' {: .4f}'.format(n).rstrip('0').rstrip('.').rstrip('-'))

def showCS(cs): # cs is an OCP.gp.gp_Ax3 object
    print("xDir ="+''.join([my_format(n) for n in cs.XDirection().XYZ().Coord()]))
    print("yDir ="+''.join([my_format(n) for n in cs.YDirection().XYZ().Coord()]))
    print("zDir ="+''.join([my_format(n) for n in cs.Direction().XYZ().Coord()]))

showCS(wp.plane.lcs)
```

adam-urbanczyk, 2020-08-06:
Indeed, you are right regarding `cq.Workplane` but it is still not possible using `cq.Workplane.workplane`. I'll adjust the description.

nopria, 2020-08-07:
You can achieve that using "copyWorkplane":

import cadquery as cq
wp = cq.Workplane("XZ").copyWorkplane(cq.Workplane(cq.Plane(origin=(0,0,0), xDir=(1,1,0), normal=(1,0,0))))

adam-urbanczyk, 2020-08-07:
Thanks, but we are now talking about really two different things. Your example does not do what `workplane().transformed(...)` would do.