CadQuery/cadquery #359: How to name a solid in order it appears in the CQ-Editor object view?

lalebarde, 2020-05-23:
Sorry if it has been already asked, I could not find an answer. Something as:

```
mysolid.set_name("some name")
print(mysolid.name)
```

[image]

jmwright, 2020-05-24:
I can't remember for sure, but I think CQ-editor respects the label property, as did the FreeCAD workbench.

https://github.com/CadQuery/cadquery/blob/a140d1035c5ef058a5c89554ef7390873e10c370/cadquery/occ_impl/shapes.py#L234

marcus7070, 2020-05-25:
> [tag(name)](https://cadquery.readthedocs.io/en/latest/classreference.html#cadquery.CQ.tag) does not bring this. That would be nice to have.

Unfortunately the method `tag` can not do that. It tags an "object", and by that I mean the Python definition of an object, not a CadQuery model. Almost every CQ operation actually returns a new object, and the tag is just a convenient way to search backwards through the chain of objects and find one in particular. You can read more about that here: https://cadquery.readthedocs.io/en/latest/primer.html#chaining

One method to display your models with a name showing is to actually not use `show_object`. When your script finishes without any calls to `show_object`, CQ-editor finds any `cadquery.Workplane` objects and displays them using the name you gave them in your script.
[image]

adam-urbanczyk, 2020-05-25:
Like so: `show_object(res,name='a')`.

lalebarde, 2020-05-25:
Both your methods work, so either `self.label = ""` or `show_object(res,name='a')`

lalebarde, 2020-05-25:
name does not work in FreeCAD which makes use of CQ v1

closed 2020-05-25 by lalebarde