CadQuery/cadquery #360: How to export parts to SVG?

lalebarde, 2020-05-24:
Hi,

I found this function in the documentation:

> **exportSvg(fileName)**
> Exports the first item on the stack as an SVG file
> For testing purposes mainly.
> **Parameters: fileName** (String, absolute path to the file) – the filename to export

I have tested with this simple example:

```
import cadquery as cq
result = cq.Workplane("XY").box(80, 60, 10).faces(">Z").workplane().hole(22)
result.exportSvg("/home/lalebarde/test/test_export_svg_solid.svg")
result.faces(">Z").exportSvg("/home/lalebarde/test/test_export_svg_face.svg")
show_object(result)
```

If I import the face in say QCAD, I get:

[image]

and for the solid:

[image]

There is only a useless 3D view. How can I export my design? I need to print plans for the faces.

lalebarde, 2020-05-24:
Right, I have [installed](https://github.com/jmwright/cadquery-freecad-module/blob/master/docs/index.md#introduction) CQ in FreeCAD and can use all FreeCAD export features.

Besides, thought it is off topic, I need to print 1:1 big pieces plans on polyester (>A0). For this purpose, I hire a professional printing service. They have 900mm or 594mm large polyester rolls, so in practice, there is no limit for the length, which is what I need for 3 meters long pieces.

I am new to FreeCAD. I could see that to get 2D plans from a 3D model, I have to use the _Drawing_ module. But the biggest it proposes is A0, requiring for my use case 3 pages, which is not desirable.

In addition, the printing service company requires pdf only.

So:

- **How could I get 3 meters long pdf exports at 1:1 scale precisely (no adjustments)?** If it is not possible, I would fall back on A0.

- **Is it possible to configure A0 overlaps so that _Drawing_ manages itself exports/prints in several pages with connection marks?** I have tried to project a selected 3m long piece in an A0 sheet with scale 1, but it has fitted it in the page.

lalebarde, 2020-05-24:
The following sequence: {FreeCAD → CQ → module Drawing A0 → Export SVG flattened → LibreOffice Draw or InkScape → Export pdf} works. If I reopen the pdf under InkScape and measure, it is equal exactly to the original parameters

Now I wonder if there is a smart process to automitize the export of tens of parts and to optimize their position on the 2D plan to have has less has possible of losses.

jmwright, 2020-05-24:
@lalebarde FreeCAD has Python scripting capability to automate some things. You can mix FreeCAD scripting with CadQuery scripting in our workbench.

https://wiki.freecadweb.org/Python_scripting_tutorial

adam-urbanczyk, 2020-05-24:
Note that this way you'll be tied to CQ1. There is `section()` function in CQ2 that allows to slice your model. My workflow for laser cutting was to export such a slice to STEP and then export this STEP to dxf using FreeCAD. Note that there is work ongoing on DXF import and eventually on DXF export directly from CQ.

BTW: you might be looking for nesting, e.g.: https://github.com/Jack000/SVGnest

lalebarde, 2020-05-25:
Hi Adam & Jeremy, thank you for your help. I could build a TechView template and use it. SVGnest is awesome and will be very usefull for me. I will also overlap several pieces.

adam-urbanczyk, 2020-08-02:
You can use the DXF export feature for exporting of 2D sections intended for CNC work (introduced by #415).

closed 2020-08-02 by adam-urbanczyk