CadQuery/cadquery #361: New Feature: Project tagged objects onto workplane

dcowden, 2020-05-24:
Since we added tagging, it would be very advantageous if you could project tagged objects into the current workplane.  This would simplify a lot of geometry-- in fact, when I draw objects in manual CAD this feature is used very, very frequently.

adam-urbanczyk, 2020-05-24:
Can you paste a screenshot?

marcus7070, 2020-05-25:
This sounds like a great use for tags, I'll add tags to... hang on. How are you projecting regular objects onto a workplane @dcowden?

dcowden, 2020-05-25:
@marcus7070 i'm not doing it yet ;)

i'd guess using GeomProjLib https://www.opencascade.com/doc/occt-7.0.0/refman/html/class_geom_proj_lib.html would be a start. 

I'll post an example where projection is helpful soon.

marcus7070, 2020-05-25:
Sounds like a great feature to have. 

I've been reading the OCC docs for half an hour now, there are heaps of options in there about projections. Some of them sound like they'd take a lot of time and effort to implement in CQ! I'm keen to see your example problem @dcowden and narrow the scope.

adam-urbanczyk, 2020-05-25:
@dcowden  usualy it is better to work at the "topology" level of OCCT. So for projection we probably want this:
https://www.opencascade.com/doc/occt-7.4.0/refman/html/class_b_rep_proj___projection.html

dcowden, 2020-05-25:
@adam-urbanczyk yes, you are right, I didn't find that one.

dcowden, 2020-05-26:
As an example, consider this odd (and contrived) example object:

[image]

This was created in Solidworks. 

The first step is very easy:
[image]

And the CQ version can be done in about 1/3 the time it even takes Solidworks to start-- so far so good:

[image]

The next step is also very easy in both tools. In solidworks, I can project the inner circle to re-use it, so that I don't need to define the hole diameter again:

[image]

Resulting in this, when extruded:
[image]

No problem for CQ-- but we do need to a work a bit harder, factoring out the diameter as a var so i can avoid duplication ( and design intent)

[image]

At this point, CQ was still faster than Solidworks. But now the trouble starts. The next step is to create that odd triangle-like boss, which in Solidworks is easy because I can project the corner points and the arc of the round boss into the current plane:  here, my design intent is to hit the corners and draw a line between them and the center, while intersecting the arc:

[image]

This is where CQ becomes clunky. The design intent is to match the corners. But to do this, i need to do the math on the locations of those vertices.  This is fairly straightforward for the corners, but really kind of a pain to calculate the intersections between the lines and the arc. Also and i think importantly, even though I _can_ compute the corners, its not desirable. In this case, its possible to compute the corner vertices quite easily. But in general, the vertices could be the result of the prior operations, which means knowing their location will not be possible.

[image]

But i've not yet computed the intersection with the arc,  ( in Solidworks i used trim-- this deficiency is out of scope for this request, but points to why 2d needs to be parametric )

At this point, its even hard to preserve the design intent: i need to refactor my design to use corner points instead of a rect in the first step-- otherwise, i have redundant information:

Though a complete different feature request, It's also worth noting that in Solidworks, I used 'extrude up to face'), which prevents me from duplicating the height of the boss above the box. in CQ, i need to do a bit more mental work to factor that out, so that the boss will reach down to the face ( the design intent here).

The 'cq-esq' description of this last feature is "create a workplan on the top face of the circular boss. Then, make a boss that intersects the two left most vertices of the bottom face ( obj.tagged("top").vertices("<Y"),  projected into my workplane, and the origin

adam-urbanczyk, 2020-05-27:
Clear, I think the conclusion is that we need separate sketching mode that can reference features of the parent solid/workplane/tag. I added a sketching epic to try to organize this.

michaelgale, 2020-05-27:
I'm not sure if tagging is a workflow concept that I would naturally use for complex modelling.  Not to say that I wouldn't learn how.  

However, one of CQ's magic secret sauce features is the amazing `Selector` class.  It is powerful for several reasons:
1. Arithmetic and boolean operations between Selectors
2. Chaining selectors 
3. Extending selectors with custom selector classes

Selectors let you filter down a subset of useful edges/vertices etc. not only for construction but for operations such as filleting, chamfering, etc. In future, it may help with exploiting OCCT features such as "defeaturing".  In a way, a Selector is like a tag, but a more "rule" based one.  As such it is more general purpose and doesn't need me to name it.

One use case that I use selectors for heavily is post-filleting a complex model.  I need to select edges carefully (both their location and the order of operations).  Therefore I have made custom selectors such as:
- EdgeLengthSelector - only picks edges conforming to one or more length "rules"
- VerticalEdgeSelector - like EdgeLengthSelector, but "vertical", not just perfect "|Z"
- EdgesCommonWithVertex - edges that meet at vertex
- EdgesCommonWithFace - all edges that share any vertex with a selected face
etc. etc. etc.
I've started to collect my CQ toolkit and have put it into this repo:  https://github.com/michaelgale/cq-kit  Its just a start and there's more to come including documentation.  However, its most used component are the custom `Selector` classes.  They truly are superheroes in the CQ workflow! :)

lenianiva, 2024-06-24:
Is there any update to this feature? It would be convenient if `placeSketch` inherits all the tags from a sketch