CadQuery/cadquery #362: In FreeCAD: "Error executing CQGI-compliant script. 'Workplane' object has no attribute 'text'"

lalebarde, 2020-05-25:
Hi,

I am working at making work my CQ script in FreeCAD. I get this error:

`Error executing CQGI-compliant script. 'Workplane' object has no attribute 'text'`

I have understood that the CQ module for FreeCAD is up-to-date. My script works well in CQ-Editor. What may happen please?

lalebarde, 2020-05-25:
After having added `name` parameters in my `show_object` calls, FreeCAD CQ module raises this new error: 

`Error executing CQGI-compliant script. show_object() got an unexpected keyword argument 'name'`

This makes me think again it does not benefit of the last CQ version. My FreeCAD deployment has been built from `FreeCAD_0.18-16146-Linux-Conda_Py3Qt5_glibc2.12-x86_64.AppImage`. This may be an explanation ?

jmwright, 2020-05-25:
Please post future issues related to the workbench on that repo: https://github.com/jmwright/cadquery-freecad-module

The workbench is only updated for the latest version in the CadQuery 1.x line. Its not compatible with CadQuery 2.0. You'll have to use CQ-editor for CadQuery 2.0 scripts.

lalebarde, 2020-05-25:
Alright Jeremy. Thanks for the tip. But I need both CQ2 and FreeCAD export capabilities with TechDraw. How can I export to pdf from CQ2 and CQ-Editor?

lalebarde, 2020-05-25:
- Have you some date to integrate CQ2 in FreeCAD?

- As a workaround, I may add an abstract layer with a CQ1/2 flag, is there a detailed comparison between CQ1 & 2 somewhere?

jmwright, 2020-05-25:
You can look at [this comment](https://github.com/jmwright/cadquery-freecad-module/issues/146#issuecomment-579750696) for a potential workaround to display CQ 2.0 objects. Issue #363 would be the only problem left if you can get that workaround to work.

lalebarde, 2020-05-25:
I am disappointed to not have read carefully your [warning](https://github.com/jmwright/cadquery-freecad-module) : "_This workbench is not compatible with CadQuery 2.x_". I don't understand what to do from the workaround you point to.

I am going to try to abstract `show_object` with a CQ1/CQ2 switch and divide and conquer until I get my model built in FreeCAD. The primary idea to benefit from all the FreeCAD environment and features was really a great one.

lalebarde, 2020-05-25:
Here is how it will look:

```
# Python module cq12adaptor
# adapts show_object to both CQ1 and CQ2 depending on the option cqver = 1 or 2
# usage : from cq12adaptor import show_object12
# replace 'show_object(obj, options=dict(color='red'))'
# by    'show_object12(obj, options=dict(color='red', cqver=1))' if CQ v1 needed
# by    'show_object12(obj, options=dict(color='red', cqver=2))' if CQ v2 needed
# If cqver is not provided, it is passed without options filtering
def show_object12(obj, options=dict(), name=''):
    if dict and 'cqver' in dict:
    else:
        show_object(obj, name=name)
```

In a first approach, I am going to process only options I use. Suggestions are welcome.

jmwright, 2020-05-25:
`show_object` is implemented separately by each execution environment (FreeCAD workbench, CQ-editor, jupyter-cadquery, etc). The name parameter (I believe) is a newer addition to CQ-editor that is not supported in the workbench. Please set the label on the shape instead within your script, ie result.val().label = "MyPart" https://github.com/jmwright/cadquery-freecad-module/blob/master/Helpers.py#L54

Also, it looks like referenced models won't update automatically in TechDraw. From what I've read this morning you have to force views to manually refresh by updating the source. Not sure if that's changed in the latest version of FreeCAD, but that's what I saw in one place.

jmwright, 2020-05-25:
I was able to get TechDraw to generate a view based on a STEP file. If you can get your workflow to work correctly with a STEP file, you could export the STEP every time from your script and then update the TechDraw view from that. That way you could use CQ-editor/CQ 2.0 for model generation with FreeCAD /TechDraw for the 2D views.

lalebarde, 2020-05-25:
Thanks Jeremy, I still don't know anythink about STEP files. CQ muto is "[Output high quality CAD formats like STEP and AMF in addition to traditional STL](https://cadquery.readthedocs.io/en/latest/)", but there is nothing in the doc to use it, [File Management and Export](https://cadquery.readthedocs.io/en/latest/apireference.html#file-management-and-export) lists:

- **importers.importStep(fileName)**  Accepts a file name and loads the STEP file into a cadquery shape  :param fileName: The path and name of the STEP file to be imported

- **exporters.exportShape(shape, exportType, …)**  param shape the shape to export. it can be a shape object, or a cadquery object. If a cadquery

but the usual links to detailed documentation is missing for both. I actually cannot export as STEP. Could you provide me some information please?

lalebarde, 2020-05-25:
> set the label on the shape instead within your script, ie result.val().label = "MyPart" 

Is it a named parameter like `show_object(obj, label='')` or an option like `show_object(obj, options=dict(label=''))`?

lalebarde, 2020-05-25:
I am following both solutions in parallel, STEP and `cq12adaptor`. Here is the last:

```
def show_object12(obj, options=dict(), name=''):
    v = 2 # default
    if options and 'cqver' in options:
        if options['cqver'] == 1:
            v = 1
            for k in ['color', 'rgba']:
                options.pop(k, None)
    if v == 1:
        return show_object(obj, options=options)
    elif v == 2:
        return show_object(obj, options=options, name=name)
```
What is strange is that in FreeCAD, I get the following error:

```
Running the Python command 'CadQueryExecuteScript' failed:
Traceback (most recent call last):
  File "/home/lalebarde/.FreeCAD/Mod/cadquery_module/CQGui/Command.py", line 208, in Activated
    if result.options and result.options["rgba"]:

('rgba',)
```

while CQGUI accepts the `rgba` option as I can read in the code, and anyway, I suppress it in `show_object12` from `options` before calling `show_object`

If you have an idea to manage this error, you would be welcome.

lalebarde, 2020-05-25:
Changing `for k in ['color', 'rgba']:` to `for k in ['color']:` make it work, but my model is broken.

Here is with CQ2:

[image]

And with CQ1 (flag for handrails that make use of lofting activated, but steps are missing):

[image]

With handrails deactivated (steps are mal-formed):

[image]

So I really want to go on with the STEP solution and waiting towards a piece of information to perform this export, because I really don't want to spend tons of time making my model work in CQ v1.

jmwright, 2020-05-25:
@lalebarde You set the label on the shape object one level down in CadQuery's layers of abstraction by calling `val`.

```python
import cadquery as cq

# make a simple box
box = cq.Workplane("XY").rect(3, 5).extrude(2)

# Here's how the label is set. Notice the call to val to go one layer down in the API
box.val().label = "box_some_tag"

print(box.val().label)

show_object(box)
```
What's wrong in the CQ 2.0 screenshot? I can't tell, and I'm unable to infer your design intent.

It looks like you're doing architectural work. Have you seen the work that Yorik from the FreeCAD team is doing in BIM? https://www.patreon.com/posts/freecad-bim-news-37134049

I know nothing about architectural CAD, but he's been focused on BIM development, and maybe it will at least be interesting for you to see it.

lalebarde, 2020-05-25:
The CQ2 one is alright and built under CQ-Editor.

lalebarde, 2020-05-25:
Here is the code taking into account your insight for `label` for the case it is useful to someone:

```
def show_object12(obj, options=dict(), name=''):
    v = 2 # default
    if options and 'cqver' in options:
        if options['cqver'] == 1:
            v = 1
            for k in ['color']: #, 'rgba']:
                options.pop(k, None)
    if v == 1:
        obj.val().label = name
        return show_object(obj, options=options)
    elif v == 2:
        return show_object(obj, options=options, name=name)
```

closed 2020-05-25 by lalebarde