CadQuery/cadquery #363: In FreeCAD : "AttributeError: 'Name' object has no attribute 'elts'"

lalebarde, 2020-05-25:
I have commented temporarily `test` usage (issue [here](https://github.com/CadQuery/cadquery/issues/362)). Now, I have several `AttributeError: 'Name' object has no attribute 'elts'` but with no means to match it with my code:

```
Opened /home/lalebarde/Documents/v8.py
Traceback (most recent call last):
  File "/home/lalebarde/.FreeCAD/Mod/cadquery_module/Libs/cadquery/cadquery/cqgi.py", line 494, in visit_Assign
    for n, v in zip(left_side.elts, node.value.elts):
AttributeError: 'Name' object has no attribute 'elts'
Unable to handle assignment for node 'Name(id='col_esc', ctx=Store())'
```

and the same error for nodes:
```
'Name(id='col_poteau', ctx=Store())'
'Name(id='col_barreau', ctx=Store())'
'Name(id='poteau_central_xy', ctx=Store())'
```
`col_xxx` are color definitions as tuples e.g. `col_poteau = (230, 153, 0, 0.25)` used in e.g.:

`show_object(cette_marche, options=dict(rgba=col_esc))`

`poteau_central_xy` is a 4 points in a plane tuple: `poteau_central_xy = ((x_pc-e_pc/2, y_pc-e_pc/2),(x_pc+e_pc/2, y_pc-e_pc/2),(x_pc+e_pc/2, y_pc+e_pc/2),(x_pc-e_pc/2, y_pc+e_pc/2))` used in a `polyline`:

`cette_marche = cette_marche.union(cq.Workplane("XY").polyline(poteau_central_xy).close().extrude(-e_m-1).translate((0, 0, (i+1) * h_m +0.5)))`

Nothing blameworthy I assume? Could the _CQ-FreeCAD_ module update be broken ?

jmwright, 2020-05-25:
What operating system are you on, and what version of Python are you using inside FreeCAD? That workbench hasn't been tested much with Python 3.8.

lalebarde, 2020-05-25:
Debian Jessie, but all CAD in a conda environment. I have checked from the FreeCAD console, Python version is 3.6.7.

jmwright, 2020-05-25:
That's odd. I think elts is part of Python's ast module, and should be available.

closed 2023-03-17 by adam-urbanczyk