CadQuery/cadquery #364: CQ Model Execute Script in FreeCAD creates a new FreeCAD model instead of updating it

lalebarde, 2020-05-25:
Hi,

Say I have a CQ script imported in FreeCAD, I have executed it and I have a 3D model. Then I build a TechDraw page with some views and so on and save the FreeCAD project.

Let's modify some parameters of the CQ script and execute it again, the page of the FreeCAD project previously defined is closed. If I reload the FreeCAD project from the disk, it is here again, but that's the old FreeCAD project of course, not with the last modifications.

I would like just an update of the FreeCAD project when the CQ script is executed, in order to keep my pages and their links to the FreeCAD 3D model. Is it possible?

jmwright, 2020-05-25:
I don't know how TechDraw pages update. The workbench is only capable of displaying the 3D model in the 3D viewer, it doesn't write an FCStd file or anything. If TechDraw will update automatically from a file on disk, you could try having the CQ script write to a STEP file.

adam-urbanczyk, 2020-08-06:
I don't think that we have the needed FreeCAD knowledge.

closed 2020-08-06 by adam-urbanczyk