CadQuery/cadquery #365: How to export a CQ model to a STEP file?

lalebarde, 2020-05-25:
I don't know anything about [STEP](https://en.wikipedia.org/wiki/ISO_10303-21) files. I need it to go on with other CAD tools to print plans. CQ muto is "[Output high quality CAD formats like STEP and AMF in addition to traditional STL](https://cadquery.readthedocs.io/en/latest/)", but I could find nothing in the doc to use it, except § [File Management and Export](https://cadquery.readthedocs.io/en/latest/apireference.html#file-management-and-export) which lists:

- **importers.importStep(fileName)**  Accepts a file name and loads the STEP file into a cadquery shape  :param fileName: The path and name of the STEP file to be imported

- **exporters.exportShape(shape, exportType, …)**  param shape the shape to export. it can be a shape object, or a cadquery object. If a cadquery

but the usual links to detailed documentation are missing for both. Could you provide me some information please to export as a STEP file?

michaelgale, 2020-05-25:
Here's a minimum working export of export and import of CQ object to STEP files:
```python
import cadquery as cq
from cadquery.occ_impl.exporters import *

# make a simple box
box = cq.Workplane("XY").rect(3, 5).extrude(2)

# export the box as a STEP file
with open("box.step", "w") as fp:
    cq.exporters.exportShape(box, ExportTypes.STEP, fp)

# import the same STEP file into a new object called samebox
samebox = cq.occ_impl.importers.importStep("box.step")

```

michaelgale, 2020-05-25:
In general, the exporter/importer API interface could be cleaned up since it is a bit inconsistent and needlessly nested "down the stack".  I know some comments have been made about cleaning this up, e.g. #334.  
I would propose making file I/O a direct peer in the fluent interface, i.e.
```python
r = cq.Workplane("XY").rect(3,5).extrude(2).exportStep("box.step")
r = cq.Workplane("XY").importStep("box.step").edges().fillet(0.2)
r = r.exportStl("boxmesh.stl")
```
The idea is that export would "transparent" from the stack point of view and import would objects to the stack, e.g.
```python
bigbox = cq.Workplane("XY").importStep("box1.step").union(importStep("box2.step").translate((0,0,1)))
```
This would allow compositing bigger/complicated shapes from pre-computed solids stored on disk. This saves time since you don't have to exhaustively re-render every element in a big assembly.

jmwright, 2020-05-25:
@michaelgale Thanks for adding that information. I'm going to also reference issue #216 to help tie all these conversations about exporters together.

You can shortcut the STEP export code above a bit using code similar to what's shown below, but yeah, the exporter and importer interface needs work. There's interest in doing it, but when it gets done will depend on the amount of free time folks have to devote to it.

```python
import cadquery as cq

# make a simple box
box = cq.Workplane("XY").rect(3, 5).extrude(2)

# export the box as a STEP file
box.val().exportStep('/home/jwright/Downloads/box.step')
```

adam-urbanczyk, 2020-05-25:
Indeed, #216 is supposed to capture discussions regarding exporters API and general improvements. I absolutely agree that they need some cleanup.

adam-urbanczyk, 2020-08-06:
In the meantime the API was simplified a little bit:
```python
from cadquery import exporters
exporters.export(shape, 'out.step')
```

adam-urbanczyk, 2020-08-06:
I'm going to close this, if you have ideas on changing the export API please use #216 .

closed 2020-08-06 by adam-urbanczyk