CadQuery/cadquery #366: Center of edge

helmecke, 2020-05-25:
```python
import cadquery as cq

def printEach(self):
    return self.eachpoint(print)

cq.Workplane.printEach = printEach

for x in [1, 3, 5, 10, 100, 1000]:
    cq.Workplane("front").rect(x, 100).edges("|X").printEach()
```
I expected every x of the resulting vector to be `0.0` but sometimes it is not.

marcus7070, 2020-05-26:
When I run your code I get
```
Vector: (0.0, -50.0, 0.0)
Vector: (0.0, 50.0, 0.0)
Vector: (-1.1102230246251565e-16, -50.0, 0.0)
Vector: (1.1102230246251565e-16, 50.0, 0.0)
Vector: (1.1102230246251565e-16, -50.0, 0.0)
Vector: (-1.1102230246251565e-16, 50.0, 0.0)
Vector: (2.220446049250313e-16, -50.0, 0.0)
Vector: (-2.220446049250313e-16, 50.0, 0.0)
Vector: (0.0, -50.0, 0.0)
Vector: (0.0, 50.0, 0.0)
Vector: (-2.842170943040401e-14, -50.0, 0.0)
Vector: (2.842170943040401e-14, 50.0, 0.0)
```

Is that what you get?

Those values are what I would expect. The centres are calculated through OCC and there is a tolerance with those calculations.

helmecke, 2020-05-26:
That's the output. I just tried the `OCP` branch and values are the same. If this is _normal_ this issue can be closed.

adam-urbanczyk, 2020-05-26:
Indeed that is to be expected. `eachpoint` calculates Center of each object using `GProp_GProps` class from OCCT. This a calculation with floating point arithmetic so 0.0 is not always exactly 0.0 .

closed 2020-05-26 by adam-urbanczyk