CadQuery/cadquery #368: .workplane() doesn't seem to respect prior rotation

r4dr3fr4d, 2020-05-28:
I'm trying to create some boxes along an arc, by rotating each one against the line between it's two top vertices closest to the previous box.  I would think that the call to .center() would place the box oriented parallel to the previous one, but it seems to ignore the rotation that each one goes though at the end of the loop.  Not sure if this is somehow related to #304...

```
s = cq.selectors.StringSyntaxSelector
tests = list()
test = cq.Workplane("front")
for i in range(0,4):
    a = -5 if i else 0
    t = (20 + 1) if i else 0
    test = test.workplane().center(0, t).box(20, 20, 5, combine=False)
    #test = cq.Workplane(test.plane, test).center(0, t).box(20, 20, 5, combine=False)  # trying a different way
    v1 = test.vertices(s(">X and <Y and >Z")).val().toTuple()
    v2 = test.vertices(s("<X and <Y and >Z")).val().toTuple()
    r = (v1, v2, a)
    test = test.rotate(r[0], r[1], r[2])
    tests.append(test)

show_object(tests)
```
[image]

You can see the boxes are "staggered" along the Z, when they should form a "smooth" arc.

jmwright, 2020-05-28:
`workplane()` does not keep track of previous rotation, and calling `rotate()` on a workplane does not mutate it, it returns a copy of all items on the stack rotated by the specified amount.

workplane:
https://github.com/CadQuery/cadquery/blob/a140d1035c5ef058a5c89554ef7390873e10c370/cadquery/cq.py#L346

rotate:
https://github.com/CadQuery/cadquery/blob/a140d1035c5ef058a5c89554ef7390873e10c370/cadquery/cq.py#L864

r4dr3fr4d, 2020-05-28:
Hmm how is one supposed to create new objects relative to existing ones then?  Would you have any suggestions about how I could create the arc idea above?

I find it weird that .workplane() will respect previous translations, but not rotations.

r4dr3fr4d, 2020-05-28:
Thanks for the reply btw.

r4dr3fr4d, 2020-05-28:
This seems to work!
```
test = cq.Workplane("front").box(10,10,5)
for i in range(0,3):
    test = test.transformed(offset=(0,5,2.5), rotate=(10,0,0)).transformed(offset=(0,5,-2.5)).box(10,10,5)
show_object(test)
```

closed 2020-05-28 by r4dr3fr4d