CadQuery/cadquery #382: importShape mentions STL file in doc string but has no ImportTypes.STL

Shimwell, 2020-06-19:
I just noticed (while trying to import an STL file) that the importShape appears to support STL as it is mentioned in the doc string but I don't think it is actually supported in the function as there is no importType == ImportTypes.STL:

https://github.com/CadQuery/cadquery/blob/f522899ae928fbd12169bebc2ed89cd5324a9ba5/cadquery/occ_impl/importers.py#L27

Just wondering if I have misunderstood or perhaps another way of importing an STL is preferred.

jmwright, 2020-06-20:
@Shimwell It looks to me like the docstring is wrong. It seems it was written to give examples of what types of things could possibly be imported rather than what actually can currently be imported. It's misleading.

adam-urbanczyk, 2020-06-21:
@Shimwell  are you interested in STL importing or just OK with fixing the docstring?

Shimwell, 2020-06-21:
We should probably try to import STL files sometime in the future seeing as they can be exported by CQ, I guess STL import is supported by OCC. I guess this is a lot of work so I understand why the doc string contains possible future features, lets leave the docstring as is for now and see if there is demand from elsewhere for such a feature. I know you have some reworking of the importers and exports in the roadmap so it could potentially fit in with that. The file format I am really interested in is h5m geometry but that is a discussion for another time.

closed 2020-06-21 by Shimwell

jmwright, 2020-06-21:
> I am really interested in is h5m geometry but that is a discussion for another time.

@Shimwell If you have time, please open another issue outlining your need/want to use h5m, and why it might be good to have CadQuery export to it. Background info would be helpful too.

Shimwell, 2020-06-21:
> > I am really interested in is h5m geometry but that is a discussion for another time.
> 
> @Shimwell If you have time, please open another issue outlining your need/want to use h5m, and why it might be good to have CadQuery export to it. Background info would be helpful too.

Just for your interest #383