CadQuery/cadquery #385: Incorrect makeHelix output

nopria, 2020-06-23:
Method `makeHelix` gives incorrect result when the height, expressed in terms of  *number of turns*, is greater than 42:

```
import cadquery as cq

R = 50 # helix radius
N = 43 # helix turns - incorrect output for N > 42
P= 10 # helix pitch

show_object(cq.Wire.makeHelix(pitch=P, height=N*P, radius=R))
```

[image]

marcus7070, 2020-06-23:
Very interesting!

If you drop the radius to 10, I can get 48.003 turns to work, 48.004 goes crazy.

adam-urbanczyk, 2020-06-23:
Good catch @nopria ! I seems that you found some limitations of the `BRepLib.BuildCurves3d_s` call in `makeHelix`. If I use different settings `BRepLib.BuildCurves3d_s(w,1e-9,MaxSegment=1000)` it works. But there will be always an upper bound what this call can handle.

closed 2020-06-24 by jmwright