CadQuery/cadquery #390: Is Shape Tessellation Working Correctly?

jmwright, 2020-06-29:
This is with the latest master of cadquery. 

Consider the following code, which creates a CQ cube and then tessellates it.

```python
import cadquery as cq

class Component:
    height = 10
    width = 10
    length = 10
    component = None

    def __init__(self):
        self.component = cq.Workplane("XY").box(self.height, self.width, self.length)

def main():
    component = Component().component.val().tessellate(tolerance=0.001)

    print(component)

if __name__ == "__main__":
    main()
```

I end up with the following tessellation output.
```
([Vector: (-5.0, -5.0, -5.0), Vector: (-5.0, -5.0, 5.0), Vector: (-5.0, 5.0, -5.0), Vector: (-5.0, 5.0, 5.0), Vector: (5.0, -5.0, -5.0), Vector: (5.0, -5.0, 5.0), Vector: (5.0, 5.0, -5.0), Vector: (5.0, 5.0, 5.0), Vector: (-5.0, -5.0, -5.0), Vector: (5.0, -5.0, -5.0), Vector: (-5.0, -5.0, 5.0), Vector: (5.0, -5.0, 5.0), Vector: (-5.0, 5.0, -5.0), Vector: (5.0, 5.0, -5.0), Vector: (-5.0, 5.0, 5.0), Vector: (5.0, 5.0, 5.0), Vector: (-5.0, -5.0, -5.0), Vector: (-5.0, 5.0, -5.0), Vector: (5.0, -5.0, -5.0), Vector: (5.0, 5.0, -5.0), Vector: (-5.0, -5.0, 5.0), Vector: (-5.0, 5.0, 5.0), Vector: (5.0, -5.0, 5.0), Vector: (5.0, 5.0, 5.0)], 
[(2, 1, 3), (2, 3, 4), (6, 5, 7), (6, 7, 8), (12, 10, 9), (12, 9, 11), (16, 14, 13), (16, 13, 15), (20, 18, 17), (20, 17, 19), (24, 22, 21), (24, 21, 23)])
```
The vertices are the first list and the indexes are the second. Notice how the vertices are repeated even though using indexes (to define triangles based on vertex indices) is supposed to eliminate that, and the 0th index doesn't appear anywhere. I take this to mean that the indexes are maybe 1 based instead of 0 based, or maybe the tessellation algorithm simply skipped the 0th index. I don't think this is expected tessellation behavior, but wanted to check to see if I'm missing something.

adam-urbanczyk, 2020-06-29:
I did bare minimum to make the tests pass: faces are tessellated separately. The indexing in OCCT is usually 1-based so that is probably the reason. I did not look into this though.

jmwright, 2020-07-28:
Moving this conversation over to #416

closed 2020-07-28 by jmwright