CadQuery/cadquery #401: Plans with Netgen?

bragostin, 2020-07-11:
I saw that you have forked a Netgen repository here https://github.com/CadQuery/netgen
Are there any plans to integrate Netgen with Cadquery?
That could be very interesting.
I am using Netgen to mesh Cadquery models to perform linear elasticity simulations using Fenics and that works pretty well.
I have tried different meshers such as Gmsh, Tetgen and Netgen and so far Netgen is the best for my usage.

adam-urbanczyk, 2020-07-12:
Netgen has a much better surface meshing algo and I'm personally interested in using it of STL export. I opened a PR https://github.com/NGSolve/netgen/pull/63 in netgen releted to direct loading of OCCT objects from memory. So far no response from the authors. Maybe comment there that you also find it interesting?

closed 2020-07-14 by bragostin

jmwright, 2020-07-14:
@bragostin Why was this closed? From the discussion thread on the Netgen repo it sounds like you now have a path forward on integration.

bragostin, 2020-07-14:
@jmwright  I think the problem is solved, the PR is closed on Netgen Github. Or I misunderstood?

jmwright, 2020-07-14:
Ok, I guess I misunderstood. I was thinking that the plan was to do some sort of tighter integration with CQ, but you guys are just using Netgen as part of a toolchain. I see now.