CadQuery/cadquery #403: Installing cq-editor overrides a local cadquery install

winksaville, 2020-07-13:
I made a mistake, this line from the "Contributing Code" is wrong:

- Install the master branch of cq-editor, if desired: conda install -c cadquery -c conda-forge cq-editor=master

When you install cq-editor-master this way, it ends up adding the current master release of cadquery to the anaconda site-packages and overriding the local cadquery installation. I verified this by adding a line of code to both the site and local versions of rect().

local package I added:
```
        print(f'rect: winks-fork xlen={xLen} yLen={yLen}')
```
And in the site-package I added:
```
        print(f'rect: site-package xlen={xLen} yLen={yLen}')
```

I created this simple test file which displays the PYTHONPATH and sys.path and outputs the BoundingBox for the shape:
```
import os, sys
import cadquery as cq # type: ignore

try:
    user_paths = os.environ['PYTHONPATH'].split(os.pathsep)
except KeyError:
    user_paths = []

print(f'user_paths={user_paths}')
print(f'sys.path={sys.path}')

length = 90
width = 30
thickness = 0.5 
w = (
    cq.Workplane("XY")
    .rect(length, width)
    .extrude(thickness)
)
#show_object(w)
bb = w.val().BoundingBox()
print(bb.xlen, bb.ylen, bb.zlen)
```
When you run the above code **before installing** cq-editor we see `rect: winks-fork xlen=90 yLen=30`:
```
$ python plate-0.5mm.py
user_paths=[]
sys.path=['/home/wink/prgs/CadQuery/projects/plates', '/opt/anaconda/envs/cq-dev/lib/python37.zip', '/opt/anaconda/envs/cq-dev/lib/python3.7', '/opt/anaconda/envs/cq-dev/lib/python3.7/lib-dynload', '/opt/anaconda/envs/cq-dev/lib/python3.7/site-packages', '/home/wink/prgs/CadQuery/forks/cadquery']
rect: winks-fork xlen=90 yLen=30
90.0 30.0 0.5
```
And then after I **install cq-editor** we see the site-package code called `rect: site-packages xlen=90 yLen=30`:
```
$ python plate-0.5mm.py
user_paths=[]
sys.path=['/home/wink/prgs/CadQuery/projects/plates', '/opt/anaconda/envs/cq-dev/lib/python37.zip', '/opt/anaconda/envs/cq-dev/lib/python3.7', '/opt/anaconda/envs/cq-dev/lib/python3.7/lib-dynload', '/opt/anaconda/envs/cq-dev/lib/python3.7/site-packages', '/home/wink/prgs/CadQuery/forks/cadquery']
rect: site-packages xlen=90 yLen=30
90.02 30.02 0.52
```

@adam-urbanczyk, how do you have your system setup so you can test both  simultaneously?

adam-urbanczyk, 2020-07-14:
I pip installed cloned CQ repo using the editable mode: `pip install -e location/of/cq/repo`

winksaville, 2020-07-14:
After running `pip install -e .` from the repo root it installed but I got a 'ModuleNotFoundError`:
```
(cq-dev) wink@3900x:~/prgs/CadQuery/forks/CQ-editor (master)
$ pip install -e .
Obtaining file:///home/wink/prgs/CadQuery/forks/CQ-editor
Installing collected packages: CQ-editor
  Running setup.py develop for CQ-editor
Successfully installed CQ-editor
(cq-dev) wink@3900x:~/prgs/CadQuery/forks/CQ-editor (master)
$ which cq-editor
/opt/anaconda/envs/cq-dev/bin/cq-editor
(cq-dev) wink@3900x:~/prgs/CadQuery/forks/CQ-editor (master)
$ cq-editor
Traceback (most recent call last):
  File "/opt/anaconda/envs/cq-dev/bin/cq-editor", line 33, in <module>
    sys.exit(load_entry_point('CQ-editor', 'gui_scripts', 'cq-editor')())
  File "/opt/anaconda/envs/cq-dev/bin/cq-editor", line 25, in importlib_load_entry_point
    return next(matches).load()
  File "/opt/anaconda/envs/cq-dev/lib/python3.7/site-packages/importlib_metadata/__init__.py", line 105, in load
    module = import_module(match.group('module'))
  File "/opt/anaconda/envs/cq-dev/lib/python3.7/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1006, in _gcd_import
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 967, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 677, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 728, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/home/wink/prgs/CadQuery/forks/CQ-editor/cq_editor/__main__.py", line 4, in <module>
    from PyQt5.QtWidgets import QApplication
ModuleNotFoundError: No module named 'PyQt5'
```
I the same 'ModuleNotFoundError if I run `python run.py`:
```
(cq-dev) wink@3900x:~/prgs/CadQuery/forks/CQ-editor (master)
$ python run.py 
Traceback (most recent call last):
  File "run.py", line 9, in <module>
    from cq_editor.__main__ import main
  File "/home/wink/prgs/CadQuery/forks/CQ-editor/cq_editor/__main__.py", line 4, in <module>
    from PyQt5.QtWidgets import QApplication
ModuleNotFoundError: No module named 'PyQt5'
```
So it seems I'm missing dependencies, how do I install all of the dependencies except `cadquery=master`?

jmwright, 2020-07-15:
It's likely that you'll run into another issue, but you could step through and do `pip install` on the packages [here](https://github.com/CadQuery/CQ-editor/blob/master/cqgui_env.yml#L7). I would probably install pyqt first, and then keep trying to run the app to see what else is missing.

winksaville, 2020-07-15:
That will work, but the goal is to modify the instructions in the "Contributing Code" section of README.md to make it "easy" for people who wish to contribute to setup the proper environment.

A solution is to set PYTHONPATH to point to my local cadquery repo, then it works. So I could add the `Update PYTHONPATH ...` line, hightlighted below, to the "Contributing Code" section:

- Create a conda development environment with something like: conda env create -n cq-dev -f environment.yml
- Activate the new conda enviornment: conda activate cq-dev
- Install the master branch of cq-editor, if desired: conda install -c cadquery -c conda-forge cq-editor=master
- **Update PYTHONPATH to the root of your local cadquery repo:
   `export PYTHONPATH=/home/wink/prgs/CadQuery/forks/cadquery`**
- Before making any changes verify that the current tests pass. Run pytest from the root of your cadquery clone, the result should end with something like:

So after exporting PYTHONPATH I see the print statements I added to my local cadquery:

```
rect: winks-fork xlen=1118.7204795069115 yLen=1118.7204795069115
rect: winks-fork xlen=1118.7204795069115 yLen=1118.7204795069115
rect: winks-fork xlen=1034.3427821793694 yLen=1034.3427821793694
rect: winks-fork xlen=1034.3427821793694 yLen=1034.3427821793694
shell winks-fork
```

Setting PYTHONPATH isn't a great solution as you would normally be set in a .bashrc|.bash_profile and it would be "system wide" as opposed to being local to a project. A solution I came up with is to create an alias that activates the cq-dev and sets the PYTHONPATH, So maybe we could add something about setting up an alias to the instructions too:
```
alias cq-dev='conda activate cq-dev ; export PYTHONPATH="/home/wink/prgs/CadQuery/forks/cadquery"'
```
Then executing `cq-dev` activates cq-dev and exports PYTHONPATH.

Other thoughts or suggestions?

adam-urbanczyk, 2020-07-16:
I meant that you should install CQ via pip -e, not CQ-editor.

winksaville, 2020-07-16:
I'm sorry, I misread what you had written. But I'm still unsure of the steps and their order,
because it seems doing a `pip -e xxx` doesn't install dependencies. Maybe you could provide
your suggested sequence of steps, something like I've done [above](https://github.com/CadQuery/cadquery/issues/403#issuecomment-659006624)?

adam-urbanczyk, 2020-07-26:
Resolved by #409

closed 2020-07-26 by adam-urbanczyk