CadQuery/cadquery #405: Step export of shape damaged after using pushpoints()

bragostin, 2020-07-14:
Consider the following code:
```python
import cadquery as cq
from math import sin, cos, pi ,floor, sqrt, atan
import numpy as np

def pcross(self, r1=1, r2=2, fn=6, forConstruction=False):
    
    def _makeCross(center):
        pnts = []
        t = pi / fn
        c = r1 * sin(t)
        rr = r1/r2
        R = r2 * sqrt(1. + (rr * sin(t))**2)
        a = atan(rr * sin(t))
        for i in range(2*fn+1):
            if i%2==0:
                pts = [r1*cos(i * t), r1*sin(i * t)]
                pnts.append(center + cq.Vector(pts[0], pts[1], 0))
            else:
                pts = [R*cos(i*t - a), R*sin(i*t - a)]
                pnts.append(center + cq.Vector(pts[0], pts[1], 0))
                pts = [R*cos(i*t + a), R*sin(i*t + a)]
                pnts.append(center + cq.Vector(pts[0], pts[1], 0))
                
        return cq.Wire.makePolygon(pnts, forConstruction)
    
    return self.eachpoint(_makeCross, True)
cq.Workplane.pcross = pcross

NPS = 4
PH = 6.
DO = 5.
DI = 0.5
FP = 1.1

cut_out = cq.Workplane('XY').workplane().spline([(PH/2*FP, DO/2*FP), (0, DI/2/FP), (-PH/2*FP, DO/2*FP)]).close().revolve(360, axisStart=(0,0), axisEnd=(1,0))
spl = cq.Workplane('YZ').workplane(offset=-PH/2).pcross(DI/2, DO/2, NPS).extrude(PH).cut(cut_out)

xyz_pl =  [(-56.30277777777778, -26.0), (-50.36388888888889, -26.0), (-42.96944444444444, -26.0), (-37.03055555555555, -26.0)]

def s_pillar(pos):
    return spl.translate(pos).val()
pillars = cq.Workplane('YZ').pushPoints(xyz_pl).each(s_pillar)
pillars.val().exportStep('pillars.stp')

show_object(pillars)
```
All is fine in CQ-Editor:
[image]
But the ``exportStep`` of the 1st of the 4 pillars looks like this:
[image]
The middle section looks missing, which is not great for meshing.
Any idea where that might come from?

jmwright, 2020-07-14:
@bragostin What version of CQ are you running? I've got a conda environment set up including CQ-editor that's built off the latest master and I get an error related to the newer Location change.

[image]

bragostin, 2020-07-14:
I was using a master from 29 May 2020 with no execution errors, now with the latest master I get the same error as you.

adam-urbanczyk, 2020-07-14:
Below is a version that works on HEAD, note that the plugin interface changed.

```python
import cadquery as cq
from math import sin, cos, pi ,floor, sqrt, atan
import numpy as np

def pcross(self, r1=1, r2=2, fn=6, forConstruction=False):
    
    def _makeCross(loc):
        pnts = []
        t = pi / fn
        c = r1 * sin(t)
        rr = r1/r2
        R = r2 * sqrt(1. + (rr * sin(t))**2)
        a = atan(rr * sin(t))
        for i in range(2*fn+1):
            if i%2==0:
                pts = [r1*cos(i * t), r1*sin(i * t)]
                pnts.append(cq.Vector(pts[0], pts[1], 0))
            else:
                pts = [R*cos(i*t - a), R*sin(i*t - a)]
                pnts.append(cq.Vector(pts[0], pts[1], 0))
                pts = [R*cos(i*t + a), R*sin(i*t + a)]
                pnts.append(cq.Vector(pts[0], pts[1], 0))
                
        return cq.Wire.makePolygon(pnts, forConstruction).locate(loc)
    
    return self.eachpoint(_makeCross, True)

cq.Workplane.pcross = pcross

NPS = 4
PH = 6.
DO = 5.
DI = 0.5
FP = 1.1

cut_out = cq.Workplane('XY').workplane().spline([(PH/2*FP, DO/2*FP), (0, DI/2/FP), (-PH/2*FP, DO/2*FP)]).close().revolve(360, axisStart=(0,0), axisEnd=(1,0))
spl = cq.Workplane('YZ').workplane(offset=-PH/2).pcross(DI/2, DO/2, NPS).extrude(PH).cut(cut_out)

xyz_pl =  [(-56.30277777777778, -26.0), (-50.36388888888889, -26.0), (-42.96944444444444, -26.0), (-37.03055555555555, -26.0)]

def s_pillar(pos):
    return spl.translate(pos).val()
pillars = cq.Workplane('YZ').pushPoints(xyz_pl).each(s_pillar)
pillars.val().exportStep('pillars.stp')
pillars.val().exportBrep('pillars.brep')

show_object(pillars)
```

adam-urbanczyk, 2020-07-14:
@bragostin if you are using Netgen you can expor to brep as a workaround using `val().exportBrep(...)`, mporting such a file in FreeCAD results in a correct geometry.

bragostin, 2020-07-14:
@adam-urbanczyk thank you, that works perfectly, ``exportBrep()`` is really a game changer in my opinion!
Actually the ``brep`` format is also accepted by Gmsh from file.
Out of curiosity, any idea why there is this problem with the step file?

adam-urbanczyk, 2020-07-14:
Just a theory: you have on surface of revolution, maybe they are not fully supported by the STEP translator. This probably that some additional conversion/approximation is needed to be able to make a working STEP file.

bragostin, 2020-07-14:
@adam-urbanczyk I think your theory is correct, why the following revolution shape, without splines, the step file displays the correct result. More specifically it seems the problem appeared as soon as the spline starts digging inside the minimal diameter.
```python
cut_out = cq.Workplane('XY').workplane().polyline([(PH/2*FP, DO/2*FP), (PH/2*(1-PLSS), DI/2/FP)]).hLine(-PH*(1-PLSS)).polyline([(-PH/2*(1-PLSS), DI/2/FP), (-PH/2*FP, DO/2*FP)]).close().revolve(360, axisStart=(0,0), axisEnd=(1,0))
```
[image]

closed 2020-07-14 by bragostin