CadQuery/cadquery #407: Cut an external thread

marcus7070, 2020-07-17:
I've been trying to cut an external thread, but I am continually running into kernel bugs. The following code (the final `cut` call in particular) slowly chews up all my 32GB of RAM then starts paging out and I have to kill CQ-editor.

```python
import math
import cadquery as cq

def profile(base, pitch, h, extra=True):
    """
    Creates a trapezoidal wire for the cross section of the thread. If the
    cross section only goes out to the major diameter, then it doesn't cut a
    cylinder for some reason. Set extra=True to extend the cross section beyond
    the major diameter, to be certain it clears the outer surface of the base
    cylinder.
    """
    vertices = [
        base + cq.Vector(-5 / 8 * h, 0, -pitch / 8)
        , base + cq.Vector(0, 0, -7 / 16 * pitch)
    ]
    if extra:
        vertices.extend([
            base + cq.Vector(h, 0, -7 / 16 * pitch)
            , base + cq.Vector(h, 0, 7 / 16 * pitch)
        ])
    vertices.extend([
        base + cq.Vector(0, 0, 7 / 16 * pitch)
        , base + cq.Vector(-5 / 8 * h, 0, pitch / 8)
    ])
    vertices.append(vertices[0])
    return vertices

def thread_solid(pitch=3.5, major_diam=30, length=10, included_angle=60, offset=0):
    """
    Makes a solid representing a metric thread. Can be unioned to a tube with
    ID == major_diam to make an internal thread, or cut from a cylinder with
    OD == major_diam for an external thread. If offset != 0, then the profile
    is offset before sweeping, which can be used to add clearance.
    """
    h = pitch / (2 * math.tan(included_angle / 2 * math.pi / 180))
    base = cq.Vector(major_diam / 2, 0, 0)
    vertices = profile(base, pitch, h)
    profile_wire = cq.Wire.makePolygon(vertices)
    if offset:
        profile_wire = profile_wire.offset2D(offset)[0]  # offset2d returns a list, we need the one and only wire returned
    # make the helical path to sweep
    path = cq.Wire.makeHelix(pitch, length + 2 * pitch, major_diam / 2, center=cq.Vector(0, 0, -pitch))
    thread = cq.Solid.sweep(profile_wire, [], path, isFrenet=True)
    print(thread.isValid())
    return thread

thread = thread_solid(length=20)
screw = (
    cq
    .Workplane(origin=(0, 0, 5))  # make sure the thread cutting tool extends below the screw
    .circle(30 / 2)
    .extrude(10)  # make sure the thread cutting toold extends above the screw
    .cut(thread)
)
```

Screenshots always help me with these issues, so here is the `screw` before being cut:
[image]

Here is the thread that I try to cut from the screw:
[image]

Here are both objects, showing the overlap:
[image]

Has anyone got a method to either get this code working, or a method that successfully cuts external threads (using an up to date CadQuery master branch)? I know threads have been done in the past, but I'm basically copying the old CQParts method here and I can't get it to work now.

jmwright, 2020-07-17:
@marcus7070 If you extend the screw cylinder that's being cut from so that it's touching all parts of the cutting threads (top to bottom along the Z axis), does it help? I have seen the kernel get weird about cutting things like that when the cutting part extends past the part being cut from.

adam-urbanczyk, 2020-07-17:
Try the code below. Note the perturbation in diameter.
[image]

```python
import math
import cadquery as cq

def profile(base, pitch, h, extra=True):
    """
    Creates a trapezoidal wire for the cross section of the thread. If the
    cross section only goes out to the major diameter, then it doesn't cut a
    cylinder for some reason. Set extra=True to extend the cross section beyond
    the major diameter, to be certain it clears the outer surface of the base
    cylinder.
    """
    vertices = [
        base + cq.Vector(-5 / 8 * h, 0, -pitch / 8)
        , base + cq.Vector(0, 0, -7 / 16 * pitch)
    ]
    if extra:
        vertices.extend([
            base + cq.Vector(h, 0, -7 / 16 * pitch)
            , base + cq.Vector(h, 0, 7 / 16 * pitch)
        ])
    vertices.extend([
        base + cq.Vector(0, 0, 7 / 16 * pitch)
        , base + cq.Vector(-5 / 8 * h, 0, pitch / 8)
    ])
    vertices.append(vertices[0])
    return vertices

def thread_solid(pitch=3.5, major_diam=30+1e-4, length=10, included_angle=60, offset=0):
    """
    Makes a solid representing a metric thread. Can be unioned to a tube with
    ID == major_diam to make an internal thread, or cut from a cylinder with
    OD == major_diam for an external thread. If offset != 0, then the profile
    is offset before sweeping, which can be used to add clearance.
    """
    h = pitch / (2 * math.tan(included_angle / 2 * math.pi / 180))
    base = cq.Vector(major_diam / 2, 0, 0)
    vertices = profile(base, pitch, h)
    profile_wire = cq.Wire.makePolygon(vertices)
    if offset:
        profile_wire = profile_wire.offset2D(offset)[0]  # offset2d returns a list, we need the one and only wire returned
    # make the helical path to sweep
    path = cq.Wire.makeHelix(pitch, length + 2 * pitch, major_diam / 2, center=cq.Vector(0, 0, -pitch))
    thread = cq.Solid.sweep(profile_wire, [], path, isFrenet=True)
    print(thread.isValid())
    return thread

thread = thread_solid(length=20)
screw = (
    cq
    .Workplane(origin=(0, 0, -5))  # make sure the thread cutting tool extends below the screw
    .circle(30 / 2)
    .extrude(30)  # make sure the thread cutting toold extends above the screw
    .cut(thread,clean=False)
)

screw = (
    screw
    .faces('<Z').workplane(offset=-10).split(False,True)
    .faces('<Z').workplane(offset=-10).split(True)
)
```

dcowden, 2020-07-17:
for what it's worth, i've done a decent amount of coding for Onshape, which under the hood uses the Parasolid kernel.  Even for Parasolid, little tricks to avoid  boundaries conditions and/or improve performance are very common. 

For a long time, i worked with only OCC and i thought 'man, it'd be nice to use a _real_ solid kernel'. Then, when i did ONshape work for a year or so, i slowly came to accept that its just a hard problem and even the best kernels are annoying compared to business-programing i'm used to.

marcus7070, 2020-07-17:
Fantastic, thankyou all. 

When I finish the rest of this model (and print the physical part) I might write up some details and share it on the mailing list. There are a lot of work-arounds and tricks here and I would like to document them.

closed 2020-07-17 by marcus7070