CadQuery/cadquery #416: Broken exports?

Peque, 2020-07-28:
As per https://github.com/CadQuery/cadquery/pull/111, the exported files (TJS or AMF) may be broken after the OCP integration.

Doing a Git bisect:

- Good:
  - 1491e0ccf446a9cbeb4a6d7b19f0922918ba648d
  - 30d6d0b9b342da7caf296dc2aa6321d531cd5d98
- Possibly bad (skipped since I could not build docs):
  - 98fa16a0fd3ba0a324eb10518fe3491134a10097
  - 4ce8fe0a7a2ebcbe1877d9be49ac9ffa816e41fd
- Bad:
  - 8d4ff71c770d6b211291dc41197095dc888896da

The exported TJS of this part:

```python
result = cadquery.Workplane("front").box(2.0, 2.0, 0.5)
```

Used to look like this:

```json
{
    "metadata" :
    {
        "formatVersion" : 3,
        "generatedBy"   : "ParametricParts",
        "vertices"      : 24,
        "faces"         : 12,
        "normals"       : 0,
        "colors"        : 0,
        "uvs"           : 0,
        "materials"     : 1,
        "morphTargets"  : 0
    },

    "scale" : 1.0,

    "materials": [    {
    "DbgColor" : 15658734,
    "DbgIndex" : 0,
    "DbgName" : "Material",
    "colorAmbient" : [0.0, 0.0, 0.0],
    "colorDiffuse" : [0.6400000190734865, 0.10179081114814892, 0.126246120426746],
    "colorSpecular" : [0.5, 0.5, 0.5],
    "shading" : "Lambert",
    "specularCoef" : 50,
    "transparency" : 1.0,
    "vertexColors" : false
    }],

    "vertices": [-1.0, -1.0, -0.25, -1.0, -1.0, 0.25, -1.0, 1.0, 0.25, -1.0, 1.0, -0.25, 1.0, -1.0, -0.25, 1.0, -1.0, 0.25, 1.0, 1.0, 0.25, 1.0, 1.0, -0.25, -1.0, -1.0, -0.25, 1.0, -1.0, -0.25, 1.0, -1.0, 0.25, -1.0, -1.0, 0.25, -1.0, 1.0, -0.25, 1.0, 1.0, -0.25, 1.0, 1.0, 0.25, -1.0, 1.0, 0.25, -1.0, -1.0, -0.25, -1.0, 1.0, -0.25, 1.0, 1.0, -0.25, 1.0, -1.0, -0.25, -1.0, -1.0, 0.25, -1.0, 1.0, 0.25, 1.0, 1.0, 0.25, 1.0, -1.0, 0.25],

    "morphTargets": [],

    "normals": [],

    "colors": [],

    "uvs": [[]],

    "faces": [0, 3, 1, 2, 0, 0, 1, 3, 0, 5, 7, 6, 0, 5, 4, 7, 0, 9, 10, 8, 0, 8, 10, 11, 0, 14, 13, 12, 0, 14, 12, 15, 0, 17, 18, 16, 0, 16, 18, 19, 0, 22, 21, 20, 0, 22, 20, 23]
}
```

And now it looks like this:

```json
{
    "metadata" :
    {
        "formatVersion" : 3,
        "generatedBy"   : "ParametricParts",
        "vertices"      : 24,
        "faces"         : 12,
        "normals"       : 0,
        "colors"        : 0,
        "uvs"           : 0,
        "materials"     : 1,
        "morphTargets"  : 0
    },

    "scale" : 1.0,

    "materials": [    {
    "DbgColor" : 15658734,
    "DbgIndex" : 0,
    "DbgName" : "Material",
    "colorAmbient" : [0.0, 0.0, 0.0],
    "colorDiffuse" : [0.6400000190734865, 0.10179081114814892, 0.126246120426746],
    "colorSpecular" : [0.5, 0.5, 0.5],
    "shading" : "Lambert",
    "specularCoef" : 50,
    "transparency" : 1.0,
    "vertexColors" : false
    }],

    "vertices": [-1.0, -1.0, -0.25, -1.0, -1.0, 0.25, -1.0, 1.0, -0.25, -1.0, 1.0, 0.25, 1.0, -1.0, -0.25, 1.0, -1.0, 0.25, 1.0, 1.0, -0.25, 1.0, 1.0, 0.25, -1.0, -1.0, -0.25, 1.0, -1.0, -0.25, -1.0, -1.0, 0.25, 1.0, -1.0, 0.25, -1.0, 1.0, -0.25, 1.0, 1.0, -0.25, -1.0, 1.0, 0.25, 1.0, 1.0, 0.25, -1.0, -1.0, -0.25, -1.0, 1.0, -0.25, 1.0, -1.0, -0.25, 1.0, 1.0, -0.25, -1.0, -1.0, 0.25, -1.0, 1.0, 0.25, 1.0, -1.0, 0.25, 1.0, 1.0, 0.25],

    "morphTargets": [],

    "normals": [],

    "colors": [],

    "uvs": [[]],

    "faces": [0, 2, 1, 3, 0, 2, 3, 4, 0, 6, 5, 7, 0, 6, 7, 8, 0, 12, 10, 9, 0, 12, 9, 11, 0, 16, 14, 13, 0, 16, 13, 15, 0, 20, 18, 17, 0, 20, 17, 19, 0, 24, 22, 21, 0, 24, 21, 23]
}
```

It seems the vertices and faces changed.

The old one is correctly loaded by Three.js, but the new one is not. Also, I tried to load an exported AMF file into an online visualization tool and it did not work.

jmwright, 2020-07-28:
I had a similar problem awhile back ([issue here](https://github.com/CadQuery/cadquery/issues/390)). The problem as I stated in that issue is not really relevant so I think I'll close it and move the conversation here. The real problem I had was that the faces were not showing correctly during rendering because many triangle indices were not in CCW order. I also wanted the normals, which I believe the current tessellator leaves out. The tests pass with the current tessellator, so I assumed I was an edge case ended up using my own tessellator which is largely a (messy) Python port of the PythonOCC tessellator. I have added that tessellator code below along with the method where I assemble the vertices and indices/triangles in case it's helpful.
```python
from OCP.BRepTools import BRepTools
from OCP.BRepMesh import BRepMesh_IncrementalMesh
from OCP.TopLoc import TopLoc_Location
from OCP.BRep import BRep_Tool
from OCP.BRepGProp import BRepGProp_Face
from OCP.gp import gp_Pnt, gp_Vec
from OCP.TopAbs import TopAbs_INTERNAL, TopAbs_REVERSED

class Face():
    number_of_triangles = 0
    number_of_invalid_triangles = 0
    number_of_coords = 0
    number_of_normals = 0
    number_of_invalid_normals = 0

    vertices = []
    triangles = []
    normals = []

class Tessellator():
    shape = None
    computed = False
    face_list = []
    vertices = []
    triangles = []
    normals = []
    tot_triangle_count = 0
    tot_vertex_count = 0
    tot_normal_count = 0

    def __init__(self, shape):
        self.shape = shape
        self.computed = False
        self.face_list = []
        self.vertices = []
        self.triangles = []
        self.normals = []
        self.tot_triangle_count = 0
        self.tot_vertex_count = 0
        self.tot_normal_count = 0

    def get_vertex_count(self):
        return self.tot_vertex_count

    def get_vertex(self, index):
        x = self.vertices[index * 3 + 0]
        y = self.vertices[index * 3 + 1]
        z = self.vertices[index * 3 + 2]

        return (x, y, z)

    def get_normal_count(self):
        return self.tot_normal_count

    def get_normal(self, index):
        x = self.normals[index * 3 + 0]
        y = self.normals[index * 3 + 1]
        z = self.normals[index * 3 + 2]

        return (x, y, z)

    def get_triangle_count(self):
        return self.tot_triangle_count

    def get_triangle_index(self, index):
        v1 = self.triangles[3 * index + 0]
        v2 = self.triangles[3 * index + 1]
        v3 = self.triangles[3 * index + 2]
[... 161 lines cut ...]
        # (vertices, triangles, normals) = shape.tessellate(tolerance)
        tess = Tessellator(shape)
        tess.tessellate(compute_edges=True, mesh_quality=mesh_quality, parallel=parallel)

        vertices = []
        indexes = []
        normals = []

        # add vertices
        for i_vert in range(tess.get_vertex_count()):
            xyz = tess.get_vertex(i_vert)
            vertices.append(Vector(*xyz))

        # add normals
        for i_norm in range(tess.get_normal_count()):
            norm = tess.get_normal(i_norm)
            normals.append(Vector(norm))

        # add triangles
        for i_tr in range(tess.get_triangle_count()):
            indexes.append(tess.get_triangle_index(i_tr))

        # return vertices, triangles, normals
        return vertices, indexes, normals
```

adam-urbanczyk, 2020-07-29:
It is a duplicate #390  - tessellation does not return correct triangle indices.

jmwright, 2020-08-03:
@adam-urbanczyk Have you already started working on fixing tessellation, or do you have a method for fixing it already figured out? All I am planning to do is adapt my code to return the vertices and triangles the same way the current tessellation code does.

adam-urbanczyk, 2020-08-03:
Could you take a look if #420 solves the issue?

Peque, 2020-08-03:
It definitely looks better! :blush: 

[image]

adam-urbanczyk, 2020-08-04:
Is this a TJS glitch @Peque ? Visibility cannot depend on the orientation.

adam-urbanczyk, 2020-08-04:
OK, tt has to be inconsistent  winding of the triangles.

adam-urbanczyk, 2020-08-04:
It should be fixed now.

jmwright, 2020-08-04:
@Peque Would you be able to test the latest commit to #420 to make sure it works for you?

Peque, 2020-08-04:
Looks great! :tada: :blush: 

[image]

PR #420 (2020-08-04): Fix tessellate
Will resolve #416
closed 2020-08-04 by jmwright