CadQuery/cadquery #421: pendingWires bug

winksaville, 2020-08-04:
John Billingsley reported a [problem with sweep]( https://groups.google.com/g/cadquery/c/UrPx8CiSdY0/m/rYtuHCR6AgAJ).

While looking into John's Billingsley problem and learning how to use sweep
I ran into a problem that I feel is a bug.

When I create two separate sketches, then put them both on the stack using
`add()` and then do a sweep(). The result is a shape that was created only
using the rectangle in s1. See the shape on the left in the screenshot below
and lines 1..22 in the code below and in [this repo](https://github.com/winksaville/cq-pendingWires-bug):
```
(cq-dev) wink@3900x:~/prgs/CadQuery/projects/pendingWires-bug
$ cat -n pendingWires-bug.py 
     1	import cadquery as cq
     2	
     3	# The path that we'll sweep
     4	path = cq.Workplane("XZ") .moveTo(0, 4) .radiusArc(endPoint=(4, 0), radius=4)
     5	#show_object(path)
     6	
     7	# Sketch 1
     8	s1 = cq.Workplane("YZ").moveTo(0, 4).rect(2, 1)
     9	#show_object(s1)
    10	
    11	# Sketch 2
    12	s2 = cq.Workplane("XY").moveTo(4, 0).circle(0.5)
    13	#show_object(s2)
    14	
    15	# BUG?: len(c.ctx.pendingWires) should equal 2 at this point, but it equals 1
    16	c = s1.add(s2)
    17	log(f"len(c.ctx.pendingWires)={len(c.ctx.pendingWires)}")
    18	#show_object(c)
    19	
    20	# This not the desired output, only the rectangle in s1 was used in the sweep
    21	result = c.sweep(path, multisection=True)
    22	show_object(result.translate((0, 0, 0)))
    23	
    24	# Clearing pending wires and looking at each sets pendingWires
    25	c.ctx.pendingWires = []
    26	r = c.each(lambda shape: shape)
    27	log(f"len(r.ctx.pendingWires)={len(r.ctx.pendingWires)}")
    28	
    29	# This is the desired result, sweep from rectangle to circle
    30	result = r.sweep(path, multisection=True)
    31	show_object(result.translate((0, 5, 0)))```
```
A work around, lines 24..31 above, is to use
[each()](https://github.com/CadQuery/cadquery/blob/cc1f8f3183a16a1d222959c5860280e7bd3259bb/cadquery/cq.py#L2077)
to look at all of the items on the stack. What `each()` does is if an item
is a `Wire` and not `forConstruction`
[it calls](https://github.com/CadQuery/cadquery/blob/cc1f8f3183a16a1d222959c5860280e7bd3259bb/cadquery/cq.py#L2121)
the method [_addPendingWire()](https://github.com/CadQuery/cadquery/blob/cc1f8f3183a16a1d222959c5860280e7bd3259bb/cadquery/cq.py#L1982).
Which appends the `Wire` to ctx.pendingWires. The result is that the desired shape is created.
See the shape on the right in the screenshot below:

[image]

marcus7070, 2020-08-05:
I think the `add` method does not do what you expect it to. If you look at the docstring:
https://github.com/CadQuery/cadquery/blob/5f67eb89dbbe8bac4fbf6cac8a06974fc34fe676/cadquery/cq.py#L353-L373

It's used to add objects to the stack, and when it says objects it mainly means objects that are subtypes of `CQ.occ_impl.Shape`. It's not a method to combine `Workplane` objects (ie. doesn't merge the parent chain, doesn't add tags from the argument, etc.).

adam-urbanczyk, 2020-08-05:
Confirmed, `add` only adds `.objects` and doe not attempt to merge `.ctx`. I'd advise to not use this method for typical modeling tasks. It is mostly useful if you want to add a  low level `cq.Shape`.

For what you are trying to achieve you should stick to a single `cq.Workplane` object. One day we'll implement #52 , but I would not count on it this year.

winksaville, 2020-08-05:
I find it much easier to create small entities and then visulaize them individually as I build up to a full object. This technique seems to work well for 3D objects. I'm surprised that I couldn't do the same with 2D objects. It does seem that support for 2D entities are lacking in functionality so #52 would be quite nice, but it seems there might be some small steps that could be taken before then.

The minimum I needed was what `each()` is doing, although I'm sure other useful information should also be gathered together to have something more general. So, is there a "simple" method that we could create that does the minimum necessary to combine 2D objects?

Also, why `each()` is [calling _addPendingWire()](https://github.com/CadQuery/cadquery/blob/cc1f8f3183a16a1d222959c5860280e7bd3259bb/cadquery/cq.py#L2121) that seems like a strange side effect to me?

adam-urbanczyk, 2020-08-05:
You can add objects to pending wires/edges by calling `toPending` explicitly. 

Each works as inteded, e.g. in the following code: `cq.Workplane().rect(1,1).vertices().circle(0.1).extrude(1)`

winksaville, 2020-08-05:
I've created [use-toPending.py]() to my repo and here is the code it works:
```
(cq-dev) wink@3900x:~/prgs/CadQuery/projects/pendingWires-bug (master)
$ cat -n use-toPending.py 
     1	import cadquery as cq
     2	
     3	def dbg(wp: cq.Workplane, name=None):
     4	    if name is None:
     5	        name = str(id(wp))
     6	    log(f"{name}: len({name}.objects)={len(wp.objects)} len({name}.ctx.pendingWires)={len(wp.ctx.pendingWires)}")
     7	    for i in range(0, len(wp.objects)):
     8	        log(f"{name} objects[{i}]: {str(id(wp.objects[i]))}")
     9	    for i in range(0, len(wp.ctx.pendingWires)):
    10	        log(f"{name} pendingWires[{i}]: {str(id(wp.ctx.pendingWires[i]))}")
    11	
    12	# The path that we'll sweep
    13	path = cq.Workplane("XZ") .moveTo(0, 4) .radiusArc(endPoint=(4, 0), radius=4)
    14	
    15	# Sketch 1, 1 object, 1 wire pending
    16	s1 = cq.Workplane("YZ").moveTo(0, 4).rect(2, 1)
    17	dbg(s1, "s1")
    18	
    19	# Sketch 2, 1 object, 1 wire pending
    20	s2 = cq.Workplane("XY").moveTo(4, 0).circle(0.5)
    21	dbg(s2, "s2")
    22	
    23	# Use add s2 to s1, 1 object, 1 wire is pending
    24	s1s2 = s1.add(s2)
    25	dbg(s1s2, "s1s2")
    26	
    27	# Now do s1s2.toPending(), 2 objects, 3 wires pending
    28	c = s1s2.toPending()
    29	dbg(c, "c")
    30	
    31	# Doing another toPending, 2 objects, 5 wires pending
    32	c5 = c.toPending()
    33	dbg(c, "c5")
    34	
    35	# We now sweep, 1 object, 0 wires pending
    36	r = c5.sweep(path, multisection=True)
    37	dbg(r, "r")
    38	show_object(r.translate((0, 0, 0)), "r")
```
But, as with `each()` it can create duplicate entires in pendingWires and pendingEdges,
as you can see in the screenshot below. Of course `sweep()` has no apparent issues but
it doesn't smell right to me:

[image]

winksaville, 2020-08-05:
> Each works as inteded, e.g. in the following code: `cq.Workplane().rect(1,1).vertices().circle(0.1).extrude(1)`

When I first read your statement above I didn't understand as `each()` was a detail in the implementation. Then I went searching for uses of `each()`, there is only one, and then found `eachpoint()`, 9 uses. And now I do see where having the side-effect appears to be deseriable. But since that side-effect isn't documented, AFAICT, it seems dangerous.

Anyway, I think the current methods should be renamed to something like `eachUpdatePendingWires()` and `eachpointUpdatePendingWires` and then `each()` and `eachpoint()` should have no side-effects.

adam-urbanczyk, 2020-08-06:
I'm afraid I don't agree. Breaking backward compatibility for nothing and adding methods that are not needed is not a good idea. 

BTW: you can achieve everything you want with e.g. `newObject` and `toPending`.

winksaville, 2020-08-06:
Do u think there is a need for a no side effect each() & eachpoint()?

My read of the docstrings didn't suggest to me there was a side effect,
should it be documented?

What about adding an optional parameter that preserves the current
behavior, but allows the no side effect behavior?

Or create new methods with no side effect, something like every() and
everypoint()?

Regarding toPending(), creating a new object is a good idea, but seems
expensive. What about an optional parameter for it that would control
assigning empty lists to pendingWires/pendingEdges before the loops?

I can come up with a couple PRs if u think it's worth it.

Just some suggestions :)