CadQuery/cadquery #425: Add a "rotateByPlanes" method

nopria, 2020-08-07:
I believe that a 3D rotation method defined through coordinate systems would be very useful. After some investigation I found that Cadquery already has all needed information and tools to easily implement such a method. Since the Plane object underlies a coordinate system, it's easy to define a 3D rotation in terms of starting and ending planes.  I report below the code (working in cq-editor, Cadquery master branch) that progammatically adds such method to Cadquery:

```
import cadquery as cq
from cadquery.occ_impl.geom import Plane
from cadquery.occ_impl.shapes import Shape

def shape_rotateByPlanes(self, toPlane :Plane, fromPlane :Plane = Plane((0,0,0),(1,0,0),(0,0,1))):
    """
    Rotates shapes by the same 3D rotation that leads coordinate system underneath Plane `fromPlane` to coincide with coordinate system underneath Plane `toPlane`. If plane origins are different, a translation component is added to the rotation.
    Under several situations it's easier to define 3D rotations by a starting and an ending coordinate system. Note that a Plane object is actually a coordinate system.  The coordinate system of a Plane is defined through 2 vectors describing the X axis and the Z axis (normal) directions of the Plane. If X axis direction is not orthogonal to Z axis, the Z-X plane is used to define the position of resulting X axis.

    :param self: the shape to be rotated in 3D.
    :type self: a Shape object
    :param toPlane: the plane underlying the coordinate system that defines the ending position of 3D rotation.
    :type toPlane: a Plane object. For example cq.Plane(origin=(0,0,0), xDir=(1,0,0), normal=(0,0,1)) underlies the global coordinate system.
    :param fromPlane: the Plane underlying the coordinate system that defines the starting position of 3D rotation. This parameter is optional and the default value is the global coordinate system.
    :type fromPlane: a Plane object. For example cq.Plane(origin=(0,0,0), xDir=(1,0,0), normal=(0,0,1)) underlies the global coordinate system.
    :returns: a 3D rotated copy of original Shape.
    """
    return self.transformShape(toPlane.rG.multiply(fromPlane.fG)) # rG is the inverse transformation matrix of a coordinate system, fG is the forward transformation matrix
cq.Shape.rotateByPlanes = shape_rotateByPlanes # adds method to cadquery

def workplane_rotateByPlanes(self, toPlane :Plane, fromPlane :Plane = Plane((0,0,0),(1,0,0),(0,0,1))) -> "Workplane":
    """
    Returns a copy of all of the items on the stack 3D rotated using Shape.rotateByPlanes method.

    :param toPlane: the Plane underlying the coordinate system that defines the ending position of 3D rotation.
    :type toPlane: a Plane object. For example cq.Plane(origin=(0,0,0), xDir=(1,0,0), normal=(0,0,1)) underlies the global coordinate system.
    :param fromPlane: the Plane underlying the coordinate system that defines the starting position of 3D rotation. This parameter is optional and the default value is the global coordinate system.
    :type fromPlane: a Plane object. For example cq.Plane(origin=(0,0,0), xDir=(1,0,0), normal=(0,0,1)) underlies the global coordinate system.
    :returns: a cq.Workplane object
    """
    return self.newObject(
        [
            o.rotateByPlanes(toPlane, fromPlane)
            if isinstance(o, Shape)
            else o
            for o in self.objects
        ]
    )
cq.Workplane.rotateByPlanes = workplane_rotateByPlanes # adds method to cadquery

pl = Plane(origin=(0,0,0), xDir=(1,0,0), normal=(0,-1,1))

show_object(cq.Workplane("XY").box(10,5,1).rotateByPlanes(pl))
```

Despite the name `rotateByPlanes`, the method adds a translation component if the origins of `fromPlane` and `toPlane` are not the same. However, since the 3D rotation method in terms of planes is the most important (and missing) functionality of the method, I believe that referring only to rotation in the name of the method is appropriate. In fact, defining a transformation between two planes with different origins that omits the translation would be counterintuitive (and more complex to code).

I apologize for any oddity in my post or in the code, I'm a non-professional programmer.

adam-urbanczyk, 2020-08-11:
Can't you use the `Location` class and its constructors for this?

nopria, 2020-08-12:
Yes you can, but why you should? Using `Location` adds complexity:

```
import cadquery as cq
obj = cq.Workplane("XY").box(10,5,1).translate((3,0,0))
pln = cq.Plane(origin=(0,0,0), xDir=(0,0,1), normal=(0,1,1))

# 3D rotate using Location
show_object(obj.val().transformShape(cq.Matrix(cq.Location(pln).wrapped.Transformation())))

# 3D rotate using just Plane
show_object(obj.val().transformShape(pln.rG))
```

And this is in the case of rotation from global coordinate system to a specified Plane. If you want to start the 3D rotation from a custom Plane, like in my method, using `Location` is even more complex.

Any method implement this functionality would boil down to a matrix multiplication, so I guess it's hard achieve it with something simpler than `toPlane.rG` (or. in case of custom starting `Plane`, `toPlane.rG.multiply(fromPlane.fG)`).

adam-urbanczyk, 2020-08-12:
> Yes you can, but why you should? Using `Location`, as far as I can understand, adds complexity:

Because it is the preferred class for representing relative and absolute location and rigid transformations and it does not add complexity. Your example is needlessly complicated:
```python
# 3D rotate using Location
show_object(obj.val().moved(cq.Location(pln))
```
We _could_ add another constructor `Location(pln1, pln2)`, but what is your use case exactly? You can already apply `Location` in relative sense.

nopria, 2020-08-13:
Of course you are right, I wasn't aware of this way of obtaining the same result. And the two planes functionality can be achieved by

`obj.val().move(cq.Location(fromPlane).inverse).move(cq.Location(toPlane))`

Maybe adding just `workplane_rotateByPlanes` would have a sense, unless you know a way to apply such a transformation to all objects in a Workplane without defining a new method.

Can you please show an example of 3D rotation applying `Location` in relative sense or I should ask it in the forum?

adam-urbanczyk, 2020-08-14:
To apply something on each element you could use `each`:
```python
.each(lambda el: el.moved(Loc))
```
`move` and `moved` apply a `Location` in a relative sense (i.e. with respect to the current location). I still don't understand why do you need to define a `Loc` with two planes. Can you give an example use case?

nopria, 2020-08-17:
That 's great, but at the end of the story maybe I'd prefer to wrap this one line code

````python
obj.each(lambda el: el.moved(cq.Location(pln)))
````

in a method, so to be able to write

````python
    obj.rotateByPlanes(pln)
````

About the reason why I'd like to be able to specify a rotation between two planes, consider the following box (in the real world the box is a compound of several solids **built on different Workplanes**):

````python
import cadquery as cq
pln1 = cq.Plane(origin=(0,0,0), xDir=(1,0,1), normal=(0,1,0))
obj = cq.Workplane(pln1).box(10,5,1)
````

and suppose you want to rotate it so that `xDir` (the longest side) goes to direction (1,1,0) and zDir (the shortest side) goes to direction (0,0,1). If you write

````python
pln2 = cq.Plane(origin=(0,0,0), xDir=(1,1,0), normal=(0,0,1))
obj.val().move(cq.Location(pln2))
````

since the rotation is relative, longest side doesn't go towards (1,1,0) and shortest side  doesn't go towards (0,0,1). You need to use 

````python
obj.val().move(cq.Location(pln1).inverse).move(cq.Location(pln2)) # rotation from pln1 to pln2
````
 to obtain the desired rotation.

adam-urbanczyk, 2020-08-17:
In this particular example you can use the code below (and apply the location in an absolute sense). I still don't get what you want to achieve. You can already start building your object on pln2 - why would you start on plane1 and then transform?
```python
obj.val().locate(cq.Location(pln2))

```

nopria, 2020-08-20:
What I want to obtain in the real world is a function producing this (a kind reinforcing bar), which may lay in any position in space:

[image]

I already did it with OpensCAD (with much less effort I must say, but for any useful purpose I need a more powerful CAD engine like OCCT and Python flexibility and power). Notice that it is slightly pitched (along the whole length).

I'm still trying to understand the logic to be followed to do the same with Cadquery.
Following your last comment I've also refactored my code to get rid of the need of explicit 3D rotations, building the objects relative to a Workplane passed as argument, but I can't find an existing method to move an object along the local Z direction (`center` and `move` allow only X,Y adjusting). Of course I can code it somehow, but this make me doubt this way is "wrong" too.