CadQuery/cadquery #428: How can I get slot2D to work?

gtdun, 2020-08-10:
_When I try something like:
import cadquery as cq
result = cq.Workplane("XY").slot2D(4, 1, 45).extrude(1)
show_object(result)
_I get the following error message:
AtributeError: 'Workplane' object has no atribute 'slot2D'

jmwright, 2020-08-10:
You probably have an older version of CadQuery. It works fine for me on the latest master. The following is one way to install CadQuery to make sure you have all the latest features.
```
conda create -n cadquery-master
conda activate cadquery-master
conda install -c cadquery -c conda-forge cadquery=master

# If you want CQ-editor too
conda install -c cadquery -c conda-forge cq-editor=master
```

jmwright, 2020-08-14:
@gtdun Is this working for you now?

gtdun, 2020-08-14:
I have been able to use the slot2D with cadquery-2 not the cadquery-master.
I am unable to get a functioning conda "cadquery-master" install but I was able to with cadquery-2 from the cadquery installation instruction web site.

I also, am unable to get a functioning conda "cq-editor" install. It is unclear to me how to accomplish this. I am not sure if a separate env needs to be setup, or a specific python version is required, or if any other prerequisites are needed.

To answer your question I can now work with slot2D but I am unable to easily view or export my cq scripts without the cq-editor that will display the script.

adam-urbanczyk, 2020-08-14:
Feel free to open an issue in the CQ-editor issue tracker. This should work:
```python
conda create -n cq-editor
conda activate cq-editor
conda install -c cadquery -c conda-forge cq-editor=master
```

adam-urbanczyk, 2020-08-17:
Closing, see https://github.com/CadQuery/CQ-editor/issues/171 for the CQ-editor related part.

closed 2020-08-17 by adam-urbanczyk