CadQuery/cadquery #431: Trivial cq.Face.makeRuledSurface(...) fails

adrianschlatter, 2020-08-11:
Ladies and Gentlemen,

while following [this](https://groups.google.com/d/msg/cadquery/5kVRpECcxAU/7no7_ja6AAAJ) example by @adam-urbanczyk to create (screw-) threads, I ran into problems with cq.Face.makeRuledSurface(...) On my system, Adams code does not produce the part shown in the post, instead it fails with an error. I think the problem is related to cq.Face.makeRuledSurface(...) and have narrowed down the problem to the following code:

```
import cadquery as cq

v0 = cq.Vector(0, 0, 0)
v1 = cq.Vector(1, 0, 0)
v2 = cq.Vector(0, 0, 1)
v3 = cq.Vector(1, 0, 1)

e0 = cq.Edge.makeLine(v0, v1)
e1 = cq.Edge.makeLine(v2, v3)

rSurf = cq.Face.makeRuledSurface(e0, e1)

show_object(rSurf)
```

Pasting this into the CQ-editor console, I get something that seems to be a line extending from the origin along the x-axis. It has an endpoint, probably at x=1. I would show a screenshot if I could find a way to include it here. Also, rSurf.isValid() returns False.

I expected that the code above produces the face enclosed when linking v0, v1, v2, v3 by straight lines.

I run

- cadquery-2.0
- on python 3.7
- in cq-editor 0.1.1
- on MacOS 10.15.5

adam-urbanczyk, 2020-08-11:
It does work for me on the latest version of CQ (based on OCCT 7.4 and OCP). Could try installing it (something like `conda install -c conda-forge -c cadquery cadquery=master`)?

adrianschlatter, 2020-08-11:
Thanks for responding. I ran said command so that I now have:

- cadquery=master
- occt-7.4.0
- ocp-7.4
- cq-editor-0.1.2

New situation:

- CQ-editor crashes upon start
- ipython crashes upon start
- python terminal still works

CQ-editor and ipython both encounter a problem when trying to import prompt_toolkit.eventloop.generator_to_async_generator (I can provide both back traces).

conda list prompt_toolkit says I have version 1.0.15 (from channel conda-forge) of this package. Interestingly, I also have a package named prompt-toolkit (note the dash instead of the underscore) in version 3.0.5.

What I can say regarding the actual problem regarding makeRuledSurface: When I run the code in my first post in a bare-bones python console, rSurf.isValid() returns True. I don't have any visualization of it, therefore I can't say whether the surface is as expected.

adam-urbanczyk, 2020-08-11:
Could you try in a fresh conda env?

jmwright, 2020-08-12:
@adrianschlatter I think you may need to install CQ-editor with `cq-editor=master`. Also, please start with a clean Conda environment as suggested by @adam-urbanczyk

adrianschlatter, 2020-08-12:
I have created a 2nd, fresh env for cadquery with both cadquery=cq-editor=master => this fixed it (the example above as well as @adam-urbanczyk's thread example). Thanks!

Given that I never used the (original) environment for anything but cadquery, I'm a bit surprised that it broke so easily. Is that to be expected when running the latest master commit?

jmwright, 2020-08-12:
Mixing master cadquery and regular cq-editor in the same environment seems to cause problems. I'm don't know Anaconda well enough to understand why.

jmwright, 2020-08-12:
I made a change to the docs earlier today because this has been a common question lately.

https://cadquery.readthedocs.io/en/latest/installation.html

adrianschlatter, 2020-08-15:
Gentlemen,

as far as I am concerned, you can close this issue.

best regards,
Adrian

closed 2020-08-15 by adam-urbanczyk