CadQuery/cadquery #433: Plane class initialization

nopria, 2020-08-12:
`Plane` class `__init__` method docstring says

    :raises: ValueError if the specified xDir is not orthogonal to the provided normal.

This is not true (you can initialize a `Plane` with `xDir` not orthogonal to `normal`) nor necessary (such a check would be in fact counterproductive).

Moreover, a necessary check is missing; `xDir` should not be parallel to `normal`, because initialization performs a vector cross product between `normal` and `xDir`, which raises the unintelligible error `Standard_ConstructionError: gp_Vec::Normalized() - vector has zero norm` if they are parallel. Try for example:

    cq.Plane(origin=(0,0,0),xDir=(0,1,0),normal=(0,2,0))

I believe docstring should be changed to

    :raises: ValueError if the specified xDir is parallel to the provided normal.

and `_setPlaneDir` should be changed as below (or something similar)

    def _setPlaneDir(self, xDir):
        """Set the vectors parallel to the plane, i.e. xDir and yDir"""
        xDir = Vector(xDir)
        self.xDir = xDir.normalized()
        self.yDir = self.zDir.cross(self.xDir)
        if self.yDir.Length < cls._eq_tolerance_norm:            # is cls. prefix really needed ?
            raise ValueError("xDir should not be parallel to provided normal")
        self.yDir = self.yDir.normalized()

provided that the line

    _eq_tolerance_norm = 1e-6

is added under `# equality tolerances` at the beginning of class definition.

adam-urbanczyk, 2020-08-12:
Indeed, the code below does not throw an error. We should probably fix it. Why are you interested in non-orthogonal coordinate systems? CQ is a CAD and not a linear algebra package.
```python
p = cq.Plane((0,0,0),(0,1,1),(0,0,1))
```

nopria, 2020-08-12:
First of all let me say that I updated my initial comment in order to be clearer, and the code in order to reflect what I say below.

I'm not interested in non-orthogonal coordinate systems. I agree that Cadquery coordinate systems should always be orthogonal, and the code I'm proposing has exactly such a goal; it assures that the coordinate system of a `Plane` is always orthogonal while letting the user comfortably define a `Plane` with `xDir` argument not orthogonal to `normal` argument (provided it is not parallel to `normal`).

Current function `_setPlaneDir`, called during `Plane` initialization, takes care of setting `yDir` direction of resulting `Plane`, so that it is orthogonal to `normal` direction. With the change I'm proposing it does the same with `xDir`, assuring that the initialized `Plane` `xDir` is orthogonal to `normal` argument direction, which is taken as invariant.

Passing an `xDir` argument strictly orthogonal to `normal` argument is an unnecessary burden for the user and might become complicated when the involved directions are not the simplest. For example, imagine you must define a relatively simple coordinate system having `(0,2,1)` as `normal` and such that vector `(1,0,1)` lays on the new XZ plane. If the user is coerced to pass an orthogonal `xDir` argument, he/she must do some calculations to find that the orthogonal `xDir` is `(5,-2,4)`:

cq.Plane(origin=(0,0,0), xDir=(5,-2,4), normal=(0,2,1)) # with current _setPlaneDir

Such calculations are basically the same my code internally does, letting the user define the coordinate system with what he/she already knows:

cq.Plane(origin=(0,0,0), xDir=(1,0,1), normal=(0,2,1)) # with proposed _setPlaneDir