CadQuery/cadquery #439: How to shrink pending wires to perform a shoulder on a face?

lalebarde, 2020-08-14:
Here is the target result:
[image]

obtained with:
```
import cadquery as cq
result = cq.Workplane("front").box(2.0, 2.0, 0.5)
result = result.faces(">Y").workplane().rect(1.8,0.3).extrude(0.2)
show_object(result)
```
Instead, I would like to do it with something like this:
```
import cadquery as cq
result = cq.Workplane("front").box(2.0, 2.0, 0.5)
result = result.faces(">Y").wires().toPending().shrink(0.1).extrude(0.2)
show_object(result)
```

How can I do it please?

jmwright, 2020-08-14:
You should be able to use the newer offset2D method to get a similar effect.

https://github.com/CadQuery/cadquery/blob/8c864e15607b84320a4f2749e1112f1aecd7e6f4/tests/test_cadquery.py#L3529

I did a quick check and the 2D looks fine, but when I go to extrude or cut it gets messed up.

***2D***
```python
import cadquery as cq

result = cq.Workplane("front").box(2.0, 2.0, 0.5)
result2 = result.faces(">Y").wires().toPending().offset2D(-0.1)#.extrude(0.2)
show_object(result)
show_object(result2)
```
[image]

***With Extrude***
```python
import cadquery as cq

result = cq.Workplane("front").box(2.0, 2.0, 0.5)
result = result.faces(">Y").wires().toPending().offset2D(-0.1).extrude(0.2)
show_object(result)
```
[image]

adam-urbanczyk, 2020-08-14:
You need to change the workplane orientation:
```python
import cadquery as cq

result = (
    cq.Workplane("XY")
    .box(2.0, 2.0, 0.5)
    .faces(">Y").wires().toPending().offset2D(-.1)
    .transformed(rotate=(-90,0,0)).extrude(.5)
)

show_object(result)
```
[image]

lalebarde, 2020-08-15:
That would be nice to be able to write:
```
result = (
    cq.Workplane("XY")
    .box(2.0, 2.0, 0.5)
    .faces(">Y").workplane().wires().toPending().offset2D(-.1)
    .extrude(.5)
)
```

adam-urbanczyk, 2020-08-17:
It would be nice, but it would break other things. The closes you can get currently is:
```python
result = (
    cq.Workplane("XY")
    .box(2.0, 2.0, 0.5)
    .faces(">Y").wires().toPending().offset2D(-.1)
    .end().end().workplane().extrude(.5)
)
```