CadQuery/cadquery #441: SphinxCadquery Performance In Docs

jmwright, 2020-08-19:
First off I like the live previews in the docs, and I'm not suggesting that we pull them back out. However, Firefox complains about the examples page now.

I have tested this on my 1-1/2 year old Linux laptop (Ubuntu 18.04) with a i7 processor and 8 GB of RAM, as well as on a 6 month old Windows laptop (Windows 10) with an i5 and 8 GB of RAM. Same result in Firefox 79.

When opening [this page](https://cadquery.readthedocs.io/en/latest/examples.html) on readthedocs the docs do eventually load, but Firefox thinks there is a script slowing the page down and asks if I want to kill it.

[image]

I also had an instance on my Linux laptop the first time I pulled the page up where the first two examples loaded, I started rotating the models, then they locked up while the rest of the page loaded. Since it was my first time on the page, I thought it was stalled.

It seems to be working fine for the community members who have tested it so far, but the message from Firefox is concerning.

rowanG077, 2020-08-19:
There does seem to be significant difference between Chrome and Firefox. On chrome the page is fine for me. On firefox it indeed is a lot slower.

Peque, 2020-08-19:
My tests so far:

| Works | OS | Browser | CPU | GPU | RAM |
| --- | --- | --- | --- | --- | --- |
| :heavy_check_mark: | Fedora 32 (Linux) | Mozilla Firefox 78.0.2 | i7-2670QM | GeForce GT 540M (Nouveau drivers) | 8 GB |
| :heavy_check_mark: | Fedora 32 (Linux) | Chromium 84.0.4147.89 | i7-2670QM | GeForce GT 540M (Nouveau drivers) | 8 GB |

Can't seem to notice a difference with Firefox. If anything, it feels faster than Chromium. :shrug:

Could you also share the browser versions, GPU (if any)... ? Maybe we can find a pattern. Unfortunately, I cannot try it with any other machine at this moment.

Code for the table, for the lazy like me :stuck_out_tongue_winking_eye: :

```md
| Works fine | OS | Browser | CPU | GPU | RAM |
| --- | --- | --- | --- | --- | --- |
| :heavy_check_mark: | Fedora 32 (Linux) | Mozilla Firefox 78.0.2 | i7-2670QM | GeForce GT 540M (Nouveau drivers) | 8 GB |
| :heavy_check_mark: | Fedora 32 (Linux) | Chromium 84.0.4147.89 | i7-2670QM | GeForce GT 540M (Nouveau drivers) | 8 GB |
```

jmwright, 2020-08-19:
| Works | OS | Browser | CPU | GPU | RAM |
| --- | --- | --- | --- | --- | --- |
| Slow Page Warning Sometimes | Ubuntu 18.04 64-bit | Firefox 79.0 |  i7-7600U @ 2.80GHz | GeForce 940MX (Nouveau driver) | 16 GB |
| OK | Ubuntu 18.04 64-bit | Chrome 84.0.4147.125 |  i7-7600U @ 2.80GHz | GeForce 940MX (Nouveau driver) | 16 GB |
| Slow Page Warning Every Time | Windows 10 64-bit | Firefox 79.0 |  i5-9300H @ 2.40GHz | GeForce GTX 1650 | 8 GB |
| OK | Windows 10 64-bit | Chrome 84.0.4147.125 |  i5-9300H @ 2.40GHz | GeForce GTX 1650 | 8 GB |

At least one core gets pegged at 100% while the page is loading and the fans speed up on both laptops.

adam-urbanczyk, 2020-08-19:
We could increase the tolerance to 0.1 again and increase the edge detection tol in sphinxcadquery (e.g. run the docs form our fork). Maybe it'll help be enough to make firefox happy.

rowanG077, 2020-08-19:
I think that's a stop gap measure at best. It's also possible to render lazily. I checked in the perf tools and on load most time is spend in the `EdgeGeometry` constructor of three.js. If we do not perform this function on load but instead when the view port approaches a three.js canvas we can have fast load times and nice tolerance. In addition we should stop rendering views that are out of view. As it stands every single view is updated even though a user only sees one of them.

adam-urbanczyk, 2020-08-19:
Are you willing to implement such improvements @rowanG077 ?

adam-urbanczyk, 2021-04-16:
Wait with this til #281 is finished.

adam-urbanczyk, 2024-06-08:
Rendering is now based on vtk.js. Does anyone get warnings about per?