CadQuery/cadquery #443: Select edges after transformed()

bragostin, 2020-08-27:
I want to create a volume having champfered edges parallel to X in the transformed referential with
```python
(
pl =  [(0, 0), (0.9900000000000002, 1.7147302994931883), (0.9033974596215564, 1.7647302994931884), 
(0, 0.19999999999999996)]
gt = cq.Workplane('YZ').workplane(offset=-2).center(-1,-20.4).transformed(rotate=(-20,0,0)).polyline(pl)
.mirrorY().extrude(100).edges('|X').chamfer(0.5/6)
)
```
but it does not work, as if it was back in the non rotated workplane.
But I don't see how to do it. Any ideas?

adam-urbanczyk, 2020-08-27:
Selectors work in world coordinates. You can specify arbitrary direction like so: `.edges('|(1,2,3)')`. Other option would be to build the object in XY including chamfers and rotate afterwards.

bragostin, 2020-08-28:
I tried this to get the exact vector parallel to rotated X:
```python
pl =  [(0, 0), (0.9900000000000002, 1.7147302994931883), (0.9033974596215564, 1.7647302994931884), 
(0, 0.19999999999999996)]
gt = cq.Workplane('YZ').workplane(offset=-2).center(-1,-20.4).transformed(rotate=(-20,0,0)).polyline(pl)
.mirrorY().extrude(100)
sel = '|' + str((cos(-20),0,sin(-20)))
gt = gt.edges(sel).chamfer(0.5/6)
```
but I get the error:
```
  File "/usr/lib/python3/dist-packages/pyparsing.py", line 2689, in parseImpl
    raise ParseException(instring, loc, self.errmsg, self)
pyparsing.ParseException: Expected {"not" term | except | exc term} (at char 22), (line:1, col:23)
```
If there is no other way I will do it with an afterwards rotation as you suggest.

adam-urbanczyk, 2020-08-28:
That sounds like a bug in the parser.

jmwright, 2020-08-28:
The parser doesn't seem to like the spaces in the tuple string. This gets past that error, but then still says edges must be selected.
```python
pl =  [(0, 0), (0.9900000000000002, 1.7147302994931883), (0.9033974596215564, 1.7647302994931884), 
(0, 0.19999999999999996)]
gt = (cq.Workplane('YZ').workplane(offset=-2).center(-1,-20.4).transformed(rotate=(-20,0,0)).polyline(pl)
    .mirrorY().extrude(100))
log(str((cos(-20),0,sin(-20))))
#sel = '|' + str((cos(-20),0,sin(-20)))
sel = r"|(0.40808206181339196,0,-0.9129452507276277)"
gt = gt.edges(sel).chamfer(0.5/6)
```

adam-urbanczyk, 2020-08-28:
Here you can see the tangents of all the edges:
```
Vector: (0.9396926207859084, 0.0, 0.34202014332566866)
Vector: (0.9396926207859084, 0.0, 0.34202014332566866)
Vector: (-0.2961981327260237, 0.49999999999999983, 0.8137976813493739)
Vector: (-0.2961981327260237, 0.49999999999999983, 0.8137976813493739)
Vector: (0.9396926207859084, 0.0, 0.34202014332566866)
Vector: (-0.1710100716628351, -0.866025403784445, 0.46984631039294206)
Vector: (-0.1710100716628351, -0.866025403784445, 0.46984631039294206)
Vector: (0.9396926207859084, 0.0, 0.34202014332566866)
Vector: (0.296198132726024, -0.5000000000000003, -0.8137976813493735)
Vector: (0.296198132726024, -0.5000000000000003, -0.8137976813493735)
Vector: (0.9396926207859084, 0.0, 0.34202014332566866)
Vector: (0.2961981327260236, 0.5000000000000003, -0.8137976813493737)
Vector: (0.2961981327260236, 0.5000000000000003, -0.8137976813493737)
Vector: (0.9396926207859084, 0.0, 0.34202014332566866)
Vector: (-0.17101007166282597, 0.866025403784445, 0.46984631039294544)
Vector: (-0.17101007166282597, 0.866025403784445, 0.46984631039294544)
Vector: (-0.2961981327260242, -0.49999999999999983, 0.8137976813493738)
Vector: (-0.2961981327260242, -0.49999999999999983, 0.8137976813493738)
```
I don't see anything similar to the query value.

jmwright, 2020-08-28:
I'm not sure on that, I just copied the value of `sel` that was logged out. Here's a version with my copy and paste completely removed. Same error of no edges selected.
```python
pl =  [(0, 0), (0.9900000000000002, 1.7147302994931883), (0.9033974596215564, 1.7647302994931884), 
(0, 0.19999999999999996)]
gt = (cq.Workplane('YZ').workplane(offset=-2).center(-1,-20.4).transformed(rotate=(-20,0,0)).polyline(pl)
    .mirrorY().extrude(100))
log(str((cos(-20),0,sin(-20))))
sel = '|' + str((cos(-20),0,sin(-20)))
sel = sel.replace(' ', '')
gt = gt.edges(sel).chamfer(0.5/6)
```

adam-urbanczyk, 2020-08-28:
What I mean is: `|(0.40808206181339196,0,-0.9129452507276277)` does not select anything because none of the edges has a tangent of this value. The bug in the parser is of course something we need to solve.

To print all the tangents you can use the following code:
```python
for e in gt.edges().vals(): print(e.tangentAt())
```

bragostin, 2020-08-28:
@adam-urbanczyk @jmwright 
There is progress, the conversion to radian was missing:
```python
import cadquery as cq
from math import cos, sin ,pi

pl =  [(0, 0), (0.9900000000000002, 1.7147302994931883), (0.9033974596215564, 1.7647302994931884), (0, 0.19999999999999996)]
gt = cq.Workplane('YZ').workplane(offset=-2).center(-1,-20.4).transformed(rotate=(-20,0,0)).polyline(pl).mirrorY().extrude(100)

for e in gt.edges().vals(): print(e.tangentAt())

sel = '|' + str((cos(20*pi/180), 0., sin(20*pi/180)))
sel = sel.replace(' ', '')
print("sel = ", sel)
gt = gt.edges(sel).chamfer(0.5/6)

show_object(gt)
```
yields 
```
    gt = gt.edges(sel).chamfer(0.5/6)
  File "/usr/local/lib/python3.7/dist-packages/cadquery-2.0-py3.7.egg/cadquery/cq.py", line 1190, in chamfer
    s = solid.chamfer(length, length2, edgeList)
  File "/usr/local/lib/python3.7/dist-packages/cadquery-2.0-py3.7.egg/cadquery/occ_impl/shapes.py", line 1594, in chamfer
    return self.__class__(chamfer_builder.Shape())
OCP.StdFail.StdFail_NotDone: BRep_API: command not done
```
so it is trying to do something
``sel`` is now in the list of tangents, but there are 8 tangents parallel to the rotated X instead of 6.

bragostin, 2020-08-29:
Problem solved, the following code gives the expected result, there were rounding issues in ``pl`` I believe:
```python
import cadquery as cq
from math import cos, sin, tan, pi
A = 20
pl =  [(0, 0), (1., 1.7147302994931883), (0.9033974596215564, 1.7647302994931884), (0, 0.2)]
gt = cq.Workplane('YZ').workplane(offset=-2).center(-1,-20.4).transformed(rotate=(-A,0,0)).polyline(pl).mirrorY().extrude(100)

sel = '|' + str((cos(A*pi/180), 0., sin(A*pi/180)))
sel = sel.replace(' ', '')

gt = gt.edges(sel).chamfer(0.1/2)

show_object(gt)
```
[image]

PR #445 (2020-08-31): Allow spaces in a vector literal
Will resolve #443
closed 2020-08-31 by jmwright