CadQuery/cadquery #446: Sweep produced weird result with non closed wire

bragostin, 2020-08-31:
Possible bug?
This code is supposed to create an hexagon wire of circular cross section with one missing side
```python
import cadquery as cq
from math import pi, cos, sin, acos, sqrt, tan

NPS = 6
dc = 10.
tc = 0.5
A = 180 / NPS
dcc = dc * tan(A*pi/180)

def pmcross(self, r1=1, fn=6, forConstruction=False):
        def _makeCross(loc):
            pnts = []
            t = 2*pi / fn
            R = r1/2 / sin(t)
            for i in range(fn):
                pts = [R*cos(i * t + pi/fn), R*sin(i * t + pi/fn)]
                pnts.append(cq.Vector(pts[0], pts[1], 0))
                    
            return cq.Wire.makePolygon(pnts, forConstruction).locate(loc)
        
        return self.eachpoint(_makeCross, True)
cq.Workplane.pmcross = pmcross

NPS = 6
path_h = cq.Workplane('XY').pmcross(dc, NPS)
cell_h0 = cq.Workplane('XZ').workplane(offset=-dcc/2).center(dc/2,0).circle(tc/2).rotateAboutCenter((0,0,1),90-A).sweep(path_h, transition='round')

show_object(path_h)
show_object(cell_h0)
```
but yields a ``Null TopoDS_Shape object`` instead. Yet the path and section and their orientation below look correct to me. Am I missing something obvious?
[image]

adam-urbanczyk, 2020-08-31:
This works:
```python
import cadquery as cq
from math import pi, cos, sin, acos, sqrt, tan

NPS = 6
dc = 10.
tc = 0.5
A = 180 / NPS
dcc = dc * tan(A*pi/180)

def pmcross(self, r1=1, fn=6, forConstruction=False):
        def _makeCross(loc):
            pnts = []
            t = 2*pi / fn
            R = r1/2 / sin(t)
            for i in range(fn):
                pts = [R*cos(i * t + pi/fn), R*sin(i * t + pi/fn)]
                pnts.append(cq.Vector(pts[0], pts[1], 0))
                    
            return cq.Wire.makePolygon(pnts, forConstruction).locate(loc)
        
        return self.eachpoint(_makeCross, True)
cq.Workplane.pmcross = pmcross

NPS = 6
path_h = cq.Workplane('XY').pmcross(dc, NPS)
cell_h0 = cq.Workplane('XZ').workplane(offset=-dcc/2).center(dc/2,0).transformed((0,60,0)).circle(tc/2).sweep(path_h, transition='round')

show_object(path_h)
show_object(cell_h0)
```

bragostin, 2020-08-31:
@adam-urbanczyk  it does! Thank you!
Would you recommend using ``transformed`` rather than ``rotate`` whenever possible then?

closed 2020-08-31 by bragostin

adam-urbanczyk, 2020-09-01:
In your code there were two circles in `pendingWires` - `transformed` and `rotate` have different semantics.