CadQuery/cadquery #447: Cannot choose which rotated plane to select

bragostin, 2020-09-06:
The code below is aimed at boring a hole through a double rotated cylinder from its outmost face to created a pipe:
```python
import cadquery as cq
import numpy as np
from scipy.spatial.transform import Rotation as R

L = 65.
W = 123.
EH = 6.
WT = 1.
FRO = 2.
FRI = 2.
PDL = 40
PDO = 6.35
PDI = 4

EV = cq.Workplane('XY').box(EH+2*WT, W, L).edges().chamfer(FRO)
EV_cutin = cq.Workplane('XY').box(EH, W-2*WT, L-2*WT).edges().chamfer(FRI)
EV = EV.cut(EV_cutin)

AP = (13,13,0)
EV = EV.faces('<Z').workplane(offset=-FRO).center(0, 40).transformed(AP).circle(PDO/2).extrude(PDL)

v = np.array([0, 0, -1])
r = R.from_euler('xyz', list(AP), degrees=True)
v = r.apply(v)
sel = '+' + str(tuple(v))

EV = EV.faces(sel).workplane().hole(PDI, PDL+FRO)

show_object(EV)
```
however it select the inmost face and bore the hole in the wrong direction as shown below. I cannot select the other face, apparently ``[1]`` selector does not work with arbitrarily oriented planes. Is there any obvious workaround?
[image]

adam-urbanczyk, 2020-09-06:
You probably want:
```python
sel = '>' + str(tuple(v))
```

bragostin, 2020-09-06:
@adam-urbanczyk this is interesting, when first tried ``sel = '>' + str(tuple(v))`` in my main script I got this:
[image]
that's why I used ``sel = '+' + str(tuple(v))``. But in the simplified script I showed above ``sel = '>' + str(tuple(v))`` yields the expected result:
[image]

So I guess something is wrong in my main script.

bragostin, 2020-09-06:
Update: if I add ``v = np.round(v, 1)`` which results in ``sel =  >(-0.2, 0.2, -0.9)`` instead of ``sel =  >(-0.2191855733945387, 0.22495105434386498, -0.9493970231495834)`` then I get the correct result in my main script too. There seem to be some rounding issue that makes CadQuery select the wrong plane.

adam-urbanczyk, 2020-09-06:
Can you post a reproducible example ? Note that there is no implicit rounding in CQ, so it sounds like your main script has some issue. Once selected, you can verify that orientation of your face using e.g. `val().normalAt()`

bragostin, 2020-09-06:
Yes, now if you set ``PDL = 26.`` in the code above, the issue should appear as below. Note that the issue disappears when PDL>=26. Might that be a kernel issue?
[image]

bragostin, 2020-09-06:
Possibly a clue: the wrong plane is selected when the condition ``sel = >(-0.2191855733945387, 0.22495105434386498, -0.9493970231495834)`` is met by another face. But I thought the exact face orientation should be respected.
[image]

adam-urbanczyk, 2020-09-07:
`>` and `<` do not filter on orientation they just order objects according to the projection of their center on the given axis. For the model above you can actually use
```python
.faces('<Z')
```
I checked the orientation of the face and the following selector works for me:
```python
sel = "+(0.22495105434386492,0.21918557339453917,-0.9493970231495835)"
```
Note that it is different than what you have in v:
```python
v
Out[28]: array([-0.21918557,  0.22495105, -0.94939702])
```
BTW: you don't need to construct workplane orientations by hand - you can always use the `.plane.zDir` property.

bragostin, 2020-09-07:
@adam-urbanczyk thank you, this helped me to sort this issue out finally!
``.faces('<Z')`` works indeed, but only if there are other faces further away. If I add a second straight longer cylinder, it selects the longer one.
But then, using the selector you identified, the following code works even with the longer cylinder:
```python
import cadquery as cq
import numpy as np
from scipy.spatial.transform import Rotation as R

L = 65.
W = 123.
EH = 6.
WT = 1.
PDL = 26.
PDO = 6.35
PDI = 4
DW = 40.

FRO = 2.
FRI = 2.

EV = cq.Workplane('XY').center(-0.5,0).box(EH+2*WT, W, L).edges().chamfer(FRO)
EV_cutin = cq.Workplane('XY').box(EH, W-2*WT, L-2*WT).edges().chamfer(FRI)
EV = EV.cut(EV_cutin)

AP = (13,13,0)
EV = EV.faces('<Z').workplane(offset=-FRO).center(0, DW).transformed(AP).circle(PDO/2).extrude(PDL)
EV = EV.faces('<Z[-1]').workplane(offset=-FRO).center(0, -DW).circle(PDO/2).extrude(1.5*PDL)

AP = (-13, -13, 90)
v = np.array([0, 0, -1])
r = R.from_euler('xyz', list(AP), degrees=True)
v = r.apply(v)
sel = '+' + str(tuple(v))

EV = EV.faces(sel).workplane().hole(PDI, PDL+FRO)

show_object(EV)
```
[image]

I don't really understand the reason for having to use a different rotation ``AP = (-13, -13, 90)`` for the selector vector to obtain the one you identified though, but it works, thanks a lot!

closed 2020-09-07 by bragostin