CadQuery/cadquery #448: cq.Compound(cq.Edge).makeCompound(...) ==> AttributeError: type object 'Edge' has no attribute 'IsNull'

Steve-Petrie, 2020-09-10:
Using CadQuery 2.0 (I believe) in CQ-editor (Version: 0.1.0) on debian 9 (stretch) linux.

Installed in a conda environment with python 3.7:

    root@debian1:/home/steve/apps# conda create --prefix ./cqeditor_conda python=3.7

I attach a test script (file name between the < and >):
    <test_compound.py.txt> (GitHub won't attach a .py file)

Line 95 in this script is:
   *ps_cmpnd = cq.Compound(cq.Edge).makeCompound(ps_edg_lst)*

When I run this script (Run > Render) in CQ-editor, the script fails in line 95 with error:

   **AttributeError: type object 'Edge' has no attribute 'IsNull'**

I attach a screen shot file showing the cq-editor window with the error message in red (file name between the < and >):
    <CQ-editor -- Run > Render - Screenshot from 2020-09-10 17-03-30.png>

But when I run the identical <test_compound.py> script in FreeCAD CadQuery Workbench (CadQuery 1.0.0), the script runs to successful completion.

In the CadQuery **show_object** output window:
   Using **right click > Fit all** and **right click > Standard views > Axomometric**:
      Displays a box made of 12 edges.

I attach a screen shot file showing the FreeCAD CadQuery **show_object** output (file name between the < and >):
    <FreeCAD -- CadQuery > Execute script - Screenshot from 2020-09-10 17-17-13.png>

Is this a bug in CadQuery 2.0? 

Thanks **!!**
[test_compound.py.txt](https://github.com/CadQuery/cadquery/files/5204818/test_compound.py.txt)
[image]
[image]

bragostin, 2020-09-11:
It works for me if I replace
```python
# make a Compound (12 edges of ps box).
ps_cmpnd = cq.Compound(cq.Edge).makeCompound(ps_edg_lst)

# wrap the compound.
ps_cq = cq.CQ(ps_cmpnd)

```
with
```python
ps_cq = cq.Workplane('XY').newObject(ps_edg_lst)
```

Steve-Petrie, 2020-09-11:
Hi Bruno,

Thanks for taking the time to work up a fix.

I could use the Workplane approach as you suggest, but I was hoping to avoid use of Workplanes, and do all the geometric calculations with pure 3D mathematics.

Although it seems that Workplanes are the preferred approach in CQ, they just don't seem all that intuitive to me :)

I'll take a look at the two (2) versions of CadQuery and try to figure out why 2.0 is failing.

Otherwise I'll just have to start using Workplanes, because there's no way I'm giving up on CadQuery. Love the parametric paradigm coupled with the power of OCCT ...

Best Regards,

Steve

----- Original Message ----- 
  From: Bruno Agostini 
  To: CadQuery/cadquery 
  Cc: Steve-Petrie ; Author 
  Sent: Friday, September 11, 2020 1:56 AM
  Subject: Re: [CadQuery/cadquery] cq.Compound(cq.Edge).makeCompound(...) ==> AttributeError: type object 'Edge' has no attribute 'IsNull' (#448)

  It works for me if I replace

# make a Compound (12 edges of ps box).
ps_cmpnd = cq.Compound(cq.Edge).makeCompound(ps_edg_lst)

# wrap the compound.
ps_cq = cq.CQ(ps_cmpnd)with

ps_cq = cq.Workplane('XY').newObject(ps_edg_lst)—

adam-urbanczyk, 2020-09-12:
You have:
```python
ps_cmpnd = cq.Compound(cq.Edge).makeCompound(ps_edg_lst)
```
it should be:
```python
ps_cmpnd = cq.Compound.makeCompound(ps_edg_lst)
```
Also note that there is no need (in CQ-editor) to wrap `cq.Shape` instances into `cq.Workplane` in order to call `show_object`. You can use `show_object` on `cq.Shape` or even `OCP.TopoDS.TopoDS_Shape` instances directly.

Steve-Petrie, 2020-09-13:
Hi Adam,

Thanks for your help -- works great !!

Also -- appreciate the tip about show_object.

Best Regards,

Steve
  ----- Original Message ----- 
  From: Adam Urbańczyk 
  To: CadQuery/cadquery 
  Cc: Steve-Petrie ; Author 
  Sent: Saturday, September 12, 2020 5:39 PM
  Subject: Re: [CadQuery/cadquery] cq.Compound(cq.Edge).makeCompound(...) ==> AttributeError: type object 'Edge' has no attribute 'IsNull' (#448)

  You have:

ps_cmpnd = cq.Compound(cq.Edge).makeCompound(ps_edg_lst)it should be:

ps_cmpnd = cq.Compound.makeCompound(ps_edg_lst)Also note that there is no need (in CQ-editor) to wrap cq.Shape instances into cq.Workplane in order to call show_object. You can use show_object on cq.Shape or even OCP.TopoDS.TopoDS_Shape instances directly.

adam-urbanczyk, 2020-09-13:
Alright, closing.

closed 2020-09-13 by adam-urbanczyk