CadQuery/cadquery #449: Feature request - Brep export with imprinted and merged surfaces

Shimwell, 2020-09-16:
Perhaps quite low down on the list of nice features but it would be great if the BREP support allowed an optional removal of duplicate surfaces and use of virtual surfaces (also called imprinting and merging)

adam-urbanczyk, 2020-09-16:
Just to be sure, does the following code do what you need in terms of merging?
```python
import OCP
import cadquery as cq

res1 = cq.Workplane().box(1,1,1).val().wrapped
res2 = cq.Workplane().box(2,2,2).val().wrapped

bldr = OCP.BOPAlgo.BOPAlgo_Splitter()
bldr.AddArgument(res1)
bldr.AddArgument(res2)
bldr.SetNonDestructive(True)

bldr.Perform()

im = bldr.Images()

merged = cq.Compound(bldr.Shape())

show_object(merged)
```

adam-urbanczyk, 2020-10-01:
@Shimwell any updates, does the code merge the faces correctly?

Shimwell, 2020-12-16:
Ah sorry Adam, I only just noticed this suggestion. Thanks very much. We just have to get the neutronics workflow working to test it, will report back as soon as possible.

adam-urbanczyk, 2021-01-19:
NB: this functionality might be also relevant for HLR and SVG export.

Shimwell, 2021-09-25:
Taking another look at this Brep export and I notice that exporting a cq.Shape has toOCC and allows brep file export. But wondering how to export a cq.Compound as a Brep file

adam-urbanczyk, 2021-09-25:
Not sure that I follow - `cq.Compound` is a `cq.Shape`.

Shimwell, 2021-09-26:
I'm sure I'm doing something wrong here, I can get shapes to export but not compounds. Thanks for taking a look

[image]

Jojain, 2021-09-26:
Have you seen that it's already implemented : 
https://cadquery.readthedocs.io/en/latest/classreference.html?highlight=exportbrep#cadquery.Shape.exportBrep

Shimwell, 2021-09-26:
Ah super that allows me to use ```.exportBrep``` and save a compound as a Brep file without having to import ```BRepTools```. Thanks for the hint. I get missed up sometimes with the export options available Shapes, Compounds, Assemblies, exporters.export.

adam-urbanczyk, 2021-09-27:
@Shimwell `exportBrep` is what you want here. Just to clarify, in your example you are exporting a `cq.Workplane` object and not one of `cq.Shape` type (you *call* it cq_solid though). `cq.Wokrplane` has a `toOCC` method and `cq.Shape` does not.

Shimwell, 2021-09-27:
Thanks and sorry for getting mixed up there. I'm now able to export all (workplanes, cq.Shape, cq.Compound) to a Brep file. The Brep file currently just has the one volume so they all appear to be combined so I shall keep tinkering to see if I can get them to be separate volumes in the Brep. I shall try exporting assemblies next. Thanks for the help

makeclean, 2021-09-28:
@Shimwell if you're viewing the BREP afterwards with FreeCAD you will see that it is one volume, but underlying in the BREP file it remains seperate, its just the FreeCAD can't interpret it properly.

Shimwell, 2021-09-28:
Thanks @makeclean I would never have guessed that FreecCAD views all volumes as one. I do like FreeCAD, you know me too well.

I've added the merge option as described at the top of this issue and noticed that the filesize of the merged brep is slightly smaller as expected.

makeclean, 2021-09-28:
I think its that FreeCAD hasn't been told how to interpret it, in fact there is a bit of a lack of dedicated BREP viewers in general.

shimwell, 2021-12-19:
I've just been using gmsh to inspect the resulting surfaces in a merged or unmerged brep file.

I made two extruded shapes that have a touching surface into brep files using this script and surface meshed them with gmsh

The script makes a merged and unmerged brep file from the two shapes

Then I loaded up the tools -> visibility menu option to see how many surfaces there are in each model.

It looks like the brep surface merge works for this simple model. I am going to try larger more complex models next.

Here is the script if just in case it is handy for anyone

```python
import cadquery
import OCP
import gmsh

result1 = (cadquery.Workplane("XY")
          .line(0, 1)
          .line(1, 0)
          .line(0, -.5)
          .close()
          .extrude(1))

result2 = (cadquery.Workplane("XY")
          .line(0, 1)
          .line(1, 0)
          .line(0, -.5)
          .close()
          .extrude(-1))

both_unmerged = cadquery.Compound.makeCompound(
                [a.val() for a in [result1, result2]]
            )

both_unmerged.exportBrep('unmerged_surfaces.brep')

bldr = OCP.BOPAlgo.BOPAlgo_Splitter()

for shape in [result1, result2]:
    bldr.AddArgument(shape.val().wrapped)

bldr.SetNonDestructive(True)

bldr.Perform()

bldr.Images()

both_merged = cadquery.Compound(bldr.Shape())

both_merged.exportBrep('merged_surfaces.brep')

gmsh.initialize()
gmsh.option.setNumber("General.Terminal", 1)
gmsh.model.add("my_model")
volumes = gmsh.model.occ.importShapes('unmerged_surfaces.brep')
gmsh.model.occ.synchronize()
gmsh.model.mesh.generate(2)
gmsh.write("unmerged_surfaces.msh")
gmsh.finalize()

gmsh.initialize()
gmsh.option.setNumber("General.Terminal", 1)
gmsh.model.add("my_model")
volumes = gmsh.model.occ.importShapes('merged_surfaces.brep')
gmsh.model.occ.synchronize()
gmsh.model.mesh.generate(2)
gmsh.write("merged_surfaces.msh")
gmsh.finalize()
```

then load up gmsh with terminal command ```gmsh``` and file-> the .msh files. Click tools and visiblity to see the number of surfaces in each meshed brep mesh.
[image]
[image]

shimwell, 2021-12-23:
I've been exporting a variety of geometries to Brep files, some multiple volume with Boolean operations. I noticed that sometimes the order of the volumes being input differs to the volumes in the file when reading the Brep back in with gmsh. Just wondering if there a way to read in Brep files with Cadquery, it would be helpful for figuring out which volume is which. I noticed in the docs that CQ an currently read in STP files and STL files.

adam-urbanczyk, 2021-12-23:
Try this: https://github.com/CadQuery/cadquery/blob/6b01d7865a4948086d41f15fba0832284f2404c3/cadquery/occ_impl/shapes.py#L472

shimwell, 2022-01-14:
I've been testing out the merged surface Brep in a neutronics workflow and it looks like it is working so thanks for adding this feature. [Demo video of Brep to neutronics prototype](https://paramak.readthedocs.io/en/main/)