CadQuery/cadquery #459: Cannot install CadQuery via conda CLI with Python 3.8.3

fergusonsa, 2020-09-26:
As the documentation just states Python 3.+ as a requirement, I tried to install CadQuery using the instructions for the [CQ-editor.](https://github.com/CadQuery/CQ-editor#installation) I first installed Anaconda (Anaconda3-2020.07-Windows-x86_64.exe). I then ran the specified command line.

```
(base) PS C:\Users\fergu> conda install -c cadquery -c conda-forge cq-editor=master
Collecting package metadata (current_repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.
Solving environment: failed with repodata from current_repodata.json, will retry with next repodata source.
Collecting package metadata (repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.
Solving environment: \
Found conflicts! Looking for incompatible packages.
This can take several minutes.  Press CTRL-C to abort.
failed

UnsatisfiableError: The following specifications were found
to be incompatible with the existing python installation in your environment:

Specifications:

  - cq-editor=master -> python[version='>=3.7,<3.8.0a0']

Your python: python=3.8

If python is on the left-most side of the chain, that's the version you've asked for.
When python appears to the right, that indicates that the thing on the left is somehow
not available for the python version you are constrained to. Note that conda will not
change your python version to a different minor version unless you explicitly specify
that.

The following specifications were found to be incompatible with your CUDA driver:

  - feature:/win-64::__cuda==11.0=0
  - feature:|@/win-64::__cuda==11.0=0

Your installed CUDA driver is: 11.0
```
Either the documentation should be updated accordingly, or the install procedure should be updated to be able to handle Python versions > 3.8.

I am working on a laptop with Windows 10 Home (Version	10.0.18363 Build 18363). Here are the details of the Python and Conda versions:
```
(base) PS C:\Users\fergu> python --version
Python 3.8.3
(base) PS C:\Users\fergu> conda list
# packages in environment at C:\Users\fergu\anaconda3:
#
# Name                    Version                   Build  Channel
_ipyw_jlab_nb_ext_conf    0.1.0                    py38_0
alabaster                 0.7.12                     py_0
anaconda                  2020.07                  py38_0
anaconda-client           1.7.2                    py38_0
anaconda-navigator        1.9.12                   py38_0
anaconda-project          0.8.4                      py_0
argh                      0.26.2                   py38_0
asn1crypto                1.3.0                    py38_0
astroid                   2.4.2                    py38_0
astropy                   4.0.1.post1      py38he774522_1
atomicwrites              1.4.0                      py_0
attrs                     19.3.0                     py_0
autopep8                  1.5.3                      py_0
babel                     2.8.0                      py_0
backcall                  0.2.0                      py_0
backports                 1.0                        py_2
backports.functools_lru_cache 1.6.1                      py_0
backports.shutil_get_terminal_size 1.0.0                    py38_2
backports.tempfile        1.0                        py_1
backports.weakref         1.0.post1                  py_1
bcrypt                    3.1.7            py38he774522_1
beautifulsoup4            4.9.1                    py38_0
bitarray                  1.4.0            py38he774522_0
bkcharts                  0.2                      py38_0
[... 259 lines cut ...]
werkzeug                  1.0.1                      py_0
wheel                     0.34.2                   py38_0
widgetsnbextension        3.5.1                    py38_0
win_inet_pton             1.1.0                    py38_0
win_unicode_console       0.5                      py38_0
wincertstore              0.2                      py38_0
winpty                    0.4.3                         4
wrapt                     1.11.2           py38he774522_0
xlrd                      1.2.0                      py_0
xlsxwriter                1.2.9                      py_0
xlwings                   0.19.5                   py38_0
xlwt                      1.3.0                    py38_0
xmltodict                 0.12.0                     py_0
xz                        5.2.5                h62dcd97_0
yaml                      0.2.5                he774522_0
yapf                      0.30.0                     py_0
zeromq                    4.3.2                ha925a31_2
zict                      2.0.0                      py_0
zipp                      3.1.0                      py_0
zlib                      1.2.11               h62dcd97_4
zope                      1.0                      py38_1
zope.event                4.4                      py38_0
zope.interface            4.7.1            py38he774522_0
zstd                      1.4.5                ha9fde0e_0
```

jmwright, 2020-09-26:
The CadQuery library supports Python 3.8, but CQ-editor does not quite yet. I will add a note during the next docs update that suggests using the Python 3.7 version of Anaconda when wanting to run CQ-editor too.

jmwright, 2020-09-29:
PR #462 added a warning for users about the lack of Python 3.8 support currently in CQ-editor.

closed 2020-09-29 by jmwright