CadQuery/cadquery #463: text() fontsize parameter dimensions unclear

nirvdrum, 2020-09-29:
The documentation for the `fontsize` parameter of the `text` function says "size of the font" but it's unclear how that size is defined. While the value is dimensionless, what does it refer to? The size of a particular character? Something else? I'm happy to open up a PR to improve the documentation. but I'd need a little clarification before doing so.

jmwright, 2020-10-11:
Looks like the font size might be in model units. https://old.opencascade.com/doc/occt-7.4.0/refman/html/class_font___b_rep_font.html#ae3b9cc7e8a840f817a0c15edd18e14e7

adam-urbanczyk, 2020-10-11:
I assume it is the maximum height of a letter + some margin. Here is a capital letter of size 1 on top of a 1x1 face:
[image]. 
I guess that the font size  (disregarding units) has the same meaning as in any other application (never checked this assumption though). BTW: FreeType is used for handling fonts on the CAD kernel side.

PR #813 (2021-07-19): text() docstring update
    * Remove following line from docstring (appears to be copy/paste error from extrude())

       extrude always *adds* material to a part.

    * Swap order of :param distance: and :param fontsize: to match argument order

    * Mention fontsize is in model units (fix #463)

    * distance: mention distance is for both extrude/cut

    * Update kind default in docstring as "regular" not "Normal"

    * Add missing fontPath arg description

    * Add examples of usage

      - Using an installed system font

      - Using a font at provided path (when font not installe [... cut]
commit 8dc1a5c1 (2021-07-19): text() docstring update (#813)

* docstring update text()

    * Remove following line from docstring (appears to be copy/paste error from extrude())

       extrude always *adds* material to a part.

    * Swap order of :param distance: and :param fontsize: to match argument order

    * Mention fontsize is in model units (fix #463)

    * distance: mention distance is for both extrude/cut

    * Update kind default in docstring as "regular" not "Normal"

    * Add missing fontPath arg description

    * Add examples of usage

      - Using an installed system font

      - Using a font at pr [... cut]
closed 2021-07-19 by adam-urbanczyk