CadQuery/cadquery #464: Allow creating named workplanes (like 'XY') through the fluent API

marcus7070, 2020-09-29:
Often it's easiest to create a workplane with one of the named orientations (like 'XY') and an offset in world coords. The `Workplane.workplane` method can't directly do this. I find myself continually abusing the `Workplane.copyWorkplane` method to create workplanes like so:
```python
part = part.copyWorkplane(cq.Workplane("YZ", origin=(dim_x, dim_y, dim_z)))
```
and I see I'm not the only one who's picked up this habit: https://github.com/CadQuery/cadquery/issues/357#issuecomment-670413077

Would you be interested in a PR that refactored the workplane generating code from `Workplane.__init__` into a new method on the Workplane class like:
```python
def namedWorkplane(inplane='XY', origin=(0, 0, 0)):
```
`Workplane.__init__` would keep the same arguments and just pass them on to `self.namedWorkplane` to set up the first plane.

Now my example above would become:
```python
part = part.namedWorkplane('YZ', origin=(dim_x, dim_y, dim_z))
```

Just as a side note, IMHO the `Workplane.workplane` method is focussed around making workplanes relative to a face. I like it how it is and I don't think it should be overloaded/overcomplicated to do this job as well. 

I'm happy to do this PR, just thought I should get some opinions before I go refactor the code, write tests and docs.

marcus7070, 2020-09-29:
I realised there is another option. `Workplane.copyWorkplane` could be changed to accept either an `occ_impl.Plane` or `Workplane` object. Then the above example would become
```python
part = part.copyWorkplane(cq.Plane.named('XY', (dim_x, dim_y, dim_z)))
```

Not too sure which version I like better, I would love to hear some opinions!

jmwright, 2020-09-29:
I guess I'd lean towards the `namedWorkplane` solution, but I rarely feel the need to use `copyWorkplane`, so I may be a skewed sample. I'd be interested in what @adam-urbanczyk thinks about this.

adam-urbanczyk, 2020-09-29:
Actually I'd vote for extending `workplane` to support this use case (`.workplane('XY')` ) and the one described in #357 (`.workplane((1,1,1)`).  I find this kind of API much cleaner. And there is still #379 waiting to be investigated.

marcus7070, 2020-10-01:
I'm agreeing with @adam-urbanczyk now, this is a great candidate for multimethods (#379).

jmwright, 2020-10-01:
It sounds like it's safe to start the PR then @marcus7070

jpoles1, 2022-08-13:
What ever happened to this PR? Is this still being worked on? I think the ```.workplane('XY')``` syntax proposed by adam above would be a fantastic addition for usability!

jmwright, 2022-08-16:
@jpoles1 I don't think a PR ever got created for this issue. @marcus7070 is busy with other things, so there's no ETA unless someone else wants to take the task over.