CadQuery/cadquery #468: Additional constraints

adam-urbanczyk, 2020-09-29:
Check e.g. the FreeCAD assy workbench for inspiration.

jmwright, 2020-09-30:
From the Assembly3 workbench's wiki: https://github.com/realthunder/FreeCAD_assembly3/wiki/Constraints-and-Solvers#list-of-constraints

Fusion 360 "joints": https://help.autodesk.com/view/fusion360/ENU/?guid=GUID-A83EFB3E-E7C4-4B78-A842-59069004BDC0

marcus7070, 2020-10-15:
Now that I've got some experience with the current `Axis` and `Point` constraints I'm really liking how they have been implemented so far. Previously, in more point-and-click CAD software the assembly constraints were black magic to me. I feel that in CadQuery I can be more precise with exactly what I want to constrain. 

Something that would make a lot of sense to me would be to have these methods to constrain a point:

- the existing `Point`, which would fix all three translational degrees of freedom (DoF)
- `Line`, which would fix two translational DoF but allow movement along the third
- `In-plane`, which would fix one translational DoF (z-axis of a plane) but allow movement in the other two (x & y axis)

There would be an equivalent 3 for rotation, with the current `Axis` with param=0 being a 2 DoF constraint, param=X would be a 1 DoF (I think?), and I guess you'd have to provide a `cq.Plane` and match `zDir` and `xDir` for a 3 rotational DoF constraint. 

I would find this set of constraints very easy to remember and apply. Much more so than the more user friendly but convoluted constraints you find in point-and-click CAD, like "insertion", "concentric" or SolidWorks' 4 types of distance mates to cylindrical surfaces.

marcus7070, 2021-06-15:
I would like to add a constraint that locks all three rotational axes. It's very common I make an assembly with just translated parts, they are all modelled in the same orientation (which currently requires a pair of `Axis` constraints).

closed 2022-02-15 by adam-urbanczyk