CadQuery/cadquery #469: Assembly docs

adam-urbanczyk, 2020-10-03:
Add a docs chapter about using assemblies and constrains.

marcus7070, 2020-11-15:
Carrying on from #503, I think it would be good to show in the docs how to make a brace that connects between parts in an assembly. The kind of parts where it would be really annoying to calculate yourself where the mounting faces wind up in the final assembly.

1. Add several parts to an assembly and constrain them
2. Solve
3. Select a wire from the mounting face in one part of the assembly
4. Use assy.objects["part_a"].loc to position the mounting face in it's final position
5. Repeat steps 3 & 4 for part_b
6. A simple loft to create the brace.

adam-urbanczyk, 2020-11-17:
Sounds good. I want to first work on #467 and then move to this.

jmwright, 2020-11-24:
We need to make sure to include the string selector additions from #514 when this gets done.

adam-urbanczyk, 2020-11-24:
Indeed, I was planing to use them in the docs too.

adam-urbanczyk, 2020-11-24:
BTW: after solving this we could release 2.1, what do you think @jmwright ?

jmwright, 2020-11-24:
> BTW: after solving this we could release 2.1

I think that would be great.

PR #524 (2020-11-30): Constraints docs [WIP]
This PR will resolve #469 .
closed 2020-11-30 by adam-urbanczyk