CadQuery/cadquery #470: Export/import docs

adam-urbanczyk, 2020-10-03:
Add a docs page about exporting and importing

- STEP
- STEP assy
- XML OCAF assy
- DXF
- BREP
- STL/AMF

coosadog, 2020-10-14:
Hey!  I'm new to CadQuery and I've looked everywhere for documentation or examples on how to export in STL format, or any format for that for that matter.  I can't find anything explicit!  Where can I find the answer?  Thanks!

jmwright, 2020-10-15:
@coosadog 

Here are some docs: https://cadquery.readthedocs.io/en/latest/apireference.html?highlight=export#file-management-and-export

And here's an example from the test suite: https://github.com/CadQuery/cadquery/blob/master/tests/test_exporters.py#L45

jmwright, 2020-10-15:
@coosadog Also, here's an example involving STL export of a simple thing I created last week to fix a toy for someone.
```python
import cadquery as cq
from cadquery import exporters

diam = 5.0

coupler = (cq.Workplane().circle(diam).extrude(20.0)
          .faces(">Z").workplane(invert=True).circle(1.05).cutBlind(8.0)
          .faces("<Z").workplane(invert=True).circle(0.8).cutBlind(12.0)
          .edges("%CIRCLE").chamfer(0.15))

exporters.export(coupler, "/home/jwright/Downloads/coupler.stl", exporters.ExportTypes.STL)
```

coosadog, 2020-10-15:
You are amazing and thank you!  The doc made no sense at all.  It wasn't
detailed enough.  I'm a software developer and expect more from the
documentation.  I'm sure you'll fix that.  The examples were very helpful.
I would think the examples that come with the download of the software
would have that in them, but they don't.  Thanks again for your help and
examples!

marcus7070, 2021-04-06:
Partially done in #668, still need to do the assembly stuff.

adam-urbanczyk, 2023-09-06:
Done, https://cadquery.readthedocs.io/en/latest/importexport.html#exporting-assemblies-to-step

closed 2023-09-06 by adam-urbanczyk