CadQuery/cadquery #473: Assembly Axis constraint sets the two directions opposite

marcus7070, 2020-10-10:
I was expecting the Axis constraint to align two axes. What it actually does is set them opposite, kind of like the "mate" alignment in other CAD software. Here's an example:

```python
import cadquery as cq

# create a box
part_a = cq.Workplane().box(10, 10, 10)
# cut a bit out of the top side so we can see which way is up
part_a = part_a.cut(part_a.faces(">Z").workplane().sphere(1, combine=False))
# second box
part_b = part_a.translate((15, 0, 0))

assy = cq.Assembly()
assy.add(part_a, name="a")
assy.add(part_b, name="b")
# create an Axis constraint using the two bottom faces
assy.constrain("a@faces@<Z", "b@faces@<Z", "Axis")
assy.solve()

show_object(assy)
```
box b gets flipped. 

If you select the opposite face by changing the constraint line to `assy.constrain("a@faces@<Z", "b@faces@>Z", "Axis")` then you get the expected result, both boxes facing the same direction. 

I almost posted this as a comment in #469, I'd be happy if the behaviour was just well documented. It works fine, I just wasn't expecting it.

adam-urbanczyk, 2020-10-10:
This is intentional, you can specify the angle (internal default is `pi`). Constraints are still experimental, but eventually the docs will be there.
```python
assy.constrain("a@faces@<Z", "b@faces@<Z", "Axis",param=0)
assy.solve()

show_object(assy)
```
[image]

marcus7070, 2020-10-12:
Ah, thanks @adam-urbanczyk. I tried to read the source and wound up thinking that the `param` argument was just a placeholder and hadn't been implemented yet! This can be closed if you like.

adam-urbanczyk, 2020-10-12:
OK, I'll consider this a part of #469 then.

closed 2020-10-12 by adam-urbanczyk