CadQuery/cadquery #476: VRML Export

sethhillbrand, 2020-10-12:
It would be nice if CQ could expose the OpenCascade VRML export functions.  This would greatly help integration with KiCad and Blender.

jmwright, 2020-10-13:
@sethhillbrand This link is to a PR in the OCP repo. OCP is our own set of Python bindings for the OCCT CAD kernel. https://github.com/CadQuery/OCP/pull/28

adam-urbanczyk, 2020-10-13:
@sethhillbrand are you after assemblies or "regular" solids exporting?

sethhillbrand, 2020-10-13:
@adam-urbanczyk I'm not quite sure what you mean when referring to VRML assemblies.  I'm looking for solids that we can apply the extra specular characteristics to.  There shouldn't be any external references in the VRML files (if that is what you mean)

adam-urbanczyk, 2020-10-13:
I meant: do you intend to use CQ assemblies (`cq.Assembly`) and export them or just export regular (i.e. `cq.Workplane` or `cq.Solid`) objects?

sethhillbrand, 2020-10-13:
Ah!  Got it, thanks!

We are slowly migrating our generators (https://gitlab.com/kicad/libraries/kicad-packages3D-generator) from FreeCAD+CadQuery 0.3 to CadQuery 2.0.  In the old paradigm, we always did a union operation in FreeCAD before exporting the file, so I think we'll be working with `cq.Solid` for now.

Are there additional considerations that we should be thinking of when looking at `cq.Assembly`?

jmwright, 2020-10-13:
@sethhillbrand I know it's a lot of work, but I'm glad to see your project updating to CQ 2.x.

If I remember right, you guys were generating a lot of ICs with a casing and pins. You could create the casing and pins separately and then put them together using an assembly. If you don't need the case and pins treated as separate entities, I'm wondering if it's worth the extra work on your side and the exporter side though.

adam-urbanczyk, 2020-10-13:
It should be possible to support VRML export of assemblies without too much effort it seems. The advantage now would be the ability to specify colors and in the future to specify materials. Not sure though how relevant the latter is for your use case @sethhillbrand .

sethhillbrand, 2020-10-13:
@adam-urbanczyk Thanks!  We'll take a look when this is merged in.  I think that VRML specular properties need to be given during export in OCCT.  I don't know of a way offhand of specifying these qualities in the internal BREP models or their faces.

adam-urbanczyk, 2020-10-14:
Please take a look at #482 and check if it works for you. Latest OCP build from our conda channel is required. I could not make the assembly export use colors so far (in theory it should just work with `TDocStd_Document`) so leaving it out for now.

sethhillbrand, 2020-10-14:
Thanks, I'll test it out.

adam-urbanczyk, 2020-11-11:
Any news @sethhillbrand ? Shall we work on merging #482 ?

sethhillbrand, 2020-11-18:
@adam-urbanczyk Yes, works nicely.  Thanks!

PR #482 (2020-11-19): VRML export
Will resolve #476
closed 2020-11-19 by jmwright