CadQuery/cadquery #479: Assemblies with Two Planar Constraints

jmwright, 2020-10-13:
I was attempting to put together an example so I could ask about adding offsets to planar constraints, and came across what seems like a constraint solving bug, at least on the surface.
This code works fine:
```python
import cadquery as cq
from cadquery import Vector, Color, Assembly, Location

w = 10
d = 10
h = 10

part1 = cq.Workplane().box(2*w,2*d,h)
part2 = cq.Workplane().box(w,d,2*h)

assy = (
    Assembly(part1, name="FIRST")
    .add(part2, color=Color(0,0,1,0.5), name="SECOND")
)

assy.constrain("FIRST@faces@>Z", "SECOND@faces@<Z", "Plane")
# assy.constrain("FIRST@faces@>Y", "SECOND@faces@<Y", "Plane")

assy.solve()

show_object(assy)
```
[image]
But this code ends up with with the second part inside the first with seemingly no relation to the intended relationship to the first part or the origin.
```python
import cadquery as cq
from cadquery import Vector, Color, Assembly, Location

w = 10
d = 10
h = 10

part1 = cq.Workplane().box(2*w,2*d,h)
part2 = cq.Workplane().box(w,d,2*h)

assy = (
    Assembly(part1, name="FIRST")
    .add(part2, color=Color(0,0,1,0.5), name="SECOND")
)

assy.constrain("FIRST@faces@<Z", "SECOND@faces@<Z", "Plane")
assy.constrain("FIRST@faces@>Y", "SECOND@faces@<Y", "Plane")

assy.solve()

show_object(assy)
```
[image]

I've tried multiple other variations, and each set of two similar planar constraints cause this behavior. I would expect the assembly to solve to something where no part is inside another, but maybe I'm missing something obvious that makes my constraints unsolvable.

marcus7070, 2020-10-13:
When I recreate your last example @jmwright, I consistently get the blue box up the other way.

[image]

What CadQuery commit are you using? I'm on 7f26c3b3c6b87fdeda44c1f40a264b4b06994209. 

I know there is a little bit of randomness in the solver and this example is probably a bad test, but I don't like the thought of assemblies flipping sides like that depending on which machine generates them!

marcus7070, 2020-10-13:
@adam-urbanczyk, is applying a `Plane` constraint equivalent to applying an `Axis` and then a `Point` constraint? 

ie. does this code:
```python
assy.constrain("FIRST@faces@>Z", "SECOND@faces@<Z", "Plane")
```
wind up giving the solver the same mathematical constraints as this code:
```python
assy.constrain("FIRST@faces@>Z", "SECOND@faces@<Z", "Axis")
assy.constrain("FIRST@faces@>Z", "SECOND@faces@<Z", "Point")
```

jmwright, 2020-10-13:
@marcus7070 Sorry to confuse the issue. I accidentally posted the screenshot from an earlier iteration of the script. With the script that I posted above (which you ran), I get the same result that you show in your screenshot.

adam-urbanczyk, 2020-10-14:
@marcus7070 is right, you probably wanted this:
```python
import cadquery as cq
from cadquery import Vector, Color, Assembly, Location

w = 10
d = 10
h = 10

part1 = cq.Workplane().box(2*w,2*d,h)
part2 = cq.Workplane().box(w,d,2*h)

assy = (
    Assembly(part1, name="FIRST")
    .add(part2, color=Color(0,0,1,0.5), name="SECOND")
)

assy.constrain("FIRST@faces@<Z", "SECOND@faces@<Z", "Axis")
assy.constrain("FIRST@faces@>Y", "SECOND@faces@<Y", "Axis")
assy.constrain("FIRST@edges@>Y and >Z", "SECOND@edges@>Y and >Z", "Point")

assy.solve()

show_object(assy)
```
[image]

jmwright, 2020-10-14:
I must need to shift my thinking somehow, because that's counter-intuitive to me. How would I create the following positioning of part2 relative to part1 with constraints? The image creates a bit of an optical illusion, but the camera is looking up from below.
[image]
The corners are aligned in one location. In a GUI CAD application one way to accomplish that would be to add three planar constraints. Do I need to start focusing more on axes and edges with CQ's constraint system?

adam-urbanczyk, 2020-10-14:
Maybe I used the name `Plane` in a confusing way. I'm open to suggestions on changing it. It is internally decomposed into point and axis constraints. So if two faces are selected the will be centered and their normals will have opposite orientations. This allows to quickly align parts in a simple way.

With the current constraint system you can use `Point` on the vertices and 3x `Axis` on the relevant faces to get what you intend.

jmwright, 2020-10-14:
> I'm open to suggestions on changing it.

I'm guessing that I won't be the first person caught by that, but let's not worry about changing it yet. I need to live with the new constraint system for awhile, and will shift my thinking to points and axes as that seems to be the way to set up the kinds of constraints that I would normally be using. I'm going to close this for now and if it keeps coming up among the community we can revisit changes to the plane constraint.

closed 2020-10-14 by jmwright