CadQuery/cadquery #483: Time For A New Release?

jmwright, 2020-10-14:
As discovered by @winksaville on Discord, the following line installs old dependencies like OCE and python-occ in the environment rather than the current OCP and OCCT deps.
```
conda install -c conda-forge -c cadquery cadquery=2
```
So much has changed since that was current. Should we do a new release to catch conda and everything else up to where we're at now?

marcus7070, 2020-10-14:
I don't think the assembly stuff should be put in a release until #469 is done. But IMHO the move from pythonocc to OCP is certainly worth a release. OCP has performed flawlessly for me (thanks again @adam-urbanczyk) , I think it's reliable enough for a release version.

winksaville, 2020-10-14:
I'd like a releaseASAP, but if #469 is just a few days away then waiting is fine by me. If it's a few weeks, than I'd like to suggest releasing more or less "as-is" would be preferable. My two cents.

marcus7070, 2020-10-15:
I don't know if the devs want to stick to any particular style of git workflow, but they could make a release branch? Branch off master at 5ad5fdfa3d13d51b203b9eb4e4802e5aca13ed21, which is just before the assembly commits. Cherry pick/merge some later fixes and clean up like:

- #111 
- #445 
- #442 
- #478 
- f341625e7aff119d18860b42a7125f0af467cae9
- 2661101d05684ce9de69bb40b9f52a973d79d14a

Then tag that as 2.1.

jmwright, 2020-10-15:
If we're not going to do a release soon we should remove the following installation method from the readme until we do.
```
conda install -c conda-forge -c cadquery cadquery=2
```
That release uses OCE and python-occ, and is broken/confusing given the current state of CadQuery and how far we've advanced.

jmwright, 2020-10-15:
And by "broken" I mean that if you create an environment that way and try run the test suite, you'll get errors because of the OCE/OCCT and python-occ/OCP mismatch. There were also a lot of things that were broken at the kernel level that were fixed by the upgrade to OCCT 7.4.

adam-urbanczyk, 2020-10-15:
We could just change the README for now as @jmwright stated. Why rush with releasing? I think it would be better to let the assy functionality ripe a little bit.

jmwright, 2020-10-15:
#485 changes the readme to have users install CadQuery from master. Hopefully that will cut down on confusion and give us more time before the next release.

closed 2020-10-15 by jmwright