CadQuery/cadquery #486: Exporter API doesn't allow for exporter customization

ArmoredBlood, 2020-10-15:
I wasn't sure where else to ask this question, but I'm in the middle of enhancing the SVG export feature to make the view/orientation of the model customizable so the output can be directly imported into a post processor for something like a laser cutter or cnc router.

I'm nearly completed with the enhancement, but am left wondering how you would like the end user to interact with the new functionality.

This is what I have currently for getSVG():
https://github.com/ArmoredBlood/cadquery/blob/custom-svg-orientation/cadquery/occ_impl/exporters/svg.py#L128

The views are meant to match the cq-editor and freecad basic model views, still working on that piece.

My issue is when it comes to exposing the options to the user, it seems like I need to flow the options through exporters/_init_.py, but that file will get very bloated with options if we end up enhancing the other exporters as well. It appears that my only avenue for passing arguments through are overloading exporters/_init_.py:export(), which will end up with a LOT of options for various exporters and passing them to each exporter, which feels like a sub-optimal design.

My initial thought was to modify the svg.py:exportSVG() to just svg.py:export(), accept a shape, filename, and **kwargs and pass them to getSVG(). Then the user API would be a simple:
`cq.exporters.svg.export(cqObj, 'output.svg', view='isometric', hideTriad=True)`

Another option would allow a generic config object to be passed through to each exporter, but that feels a bit indeterministic. 

I wanted to run it by you to see what your thoughts were on this before I spend any more time exposing the new options. I think moving the API from exporters/_init_.py to the individual exporter files would alleviate this concern.

adam-urbanczyk, 2020-10-16:
Maybe add this to `export` and pass to `exportSVG`?
```
svgOptions: Dict[str, Any] = {}
```

ArmoredBlood, 2020-10-17:
is that how you'd like to see it done? So the end user api would be something like this?
`cq.exporters.export(cqObject, "filename.svg", "SVG", svgOptions = { view="isometric", hideTriad=True, opts={"width": 800, "height": 240, "marginLeft": 200, "marginTop": 20}})`

in my opinion, it's a bit syntax heavy. I don't really see the value of using the init.py:export function unless it's a holdover from an earlier version. 

I also don't want to cause any trouble, so if you'd rather go the svgOptions route, I can make that happen.

adam-urbanczyk, 2020-10-17:
That'd be my proposal. Note that you don't need to specify "SVG" - it will be inferred from the filename.  I assume 99% of the users will just call `export` without the extra options. @jmwright what are your thoughts?

jmwright, 2020-10-18:
If we embrace the options method, maybe just pass a generic `options` object to the export method so there doesn't have to be a different options argument for each exporter type?

If I'm understanding everything, it seems like the most viable alternative to the options method is below, and I do like how straight-forward this is.
```
cq.exporters.svg.export(cqObj, 'output.svg', view='isometric', hideTriad=True)
```
I'm still kind of on the fence between the two methods.

adam-urbanczyk, 2020-10-18:
Thanks @jmwright , indeed using single options keyword sounds much better and is future proof. My refined proposal would be:

1. add `opts: Dict[str, Any]={}` to `export` and pass it to the underlying exporters. For now this would be `exportSVG` only, but in the future possibly to other exporters.
2. Add the new keyword arguments explicitly to `exportSVG`.
3. Do not change `exportSVG` to `export` - there is no reason to make such a breaking change.

This way a regular user can still do his thing, an advanced user can tune the SVG output either using `export` or `exportSVG`  and, last but not least, backwards compatibility is maintained and there is no risk of shadowing export with export(SVG).
 
Also note that 1 and 2 are not mutually exclusive but rather complementary. The only difference of opinions is regarding 3.

Regarding the `opts` idiom itself, it is used in e.g. scipy or seaborn so I don't think it is weird or uncommon, but no one will force the users to use it.

jmwright, 2020-10-18:
@adam-urbanczyk I think that sounds like a good plan. What are your thoughts @ArmoredBlood  ?

dcowden, 2020-10-18:
I prefer the options idiom as well, but I have two thoughts:

1. Besides the cleaner, future proof signature, I tend to like it because
it's easier to read options from a conf file, etc, and to manage values
when writing  tests

2. One problem I typically see is that the options argument is poorly
documented, because nothing in the signature is there to help.  So it's
important to make that extra little effort in the doc string to explain all
the options available

2a. As a result of both the above, It'd be nice to provide a way to get a
fully populated set of defaults. If thats done using some relativel y
verbose code, it can go a long way to documenting.

ArmoredBlood, 2020-10-18:
That's fine with me, I'll add the options dict to the export function and pass it to exportSVG.  Thanks for the input everyone!