CadQuery/cadquery #494: Summary for curated list of code cade

Irev-Dev, 2020-10-31:
Hey 👋 ,

I finally got around to putting together a list of all of the code-CAD projects that I know of [here](https://kurthutten.com/blog/curated-code-cad), and also on [Github](https://github.com/Irev-Dev/curated-code-cad).

I've put a short summary of each one, but it's hard to do them all justice. for cadquery I wrote:
CadQuery is a Python library that wraps and extends OpenCascade. It aims to let you model as "close as possible to how you’d describe the object to a human".

If you would like to give me a summary of the project, and perhaps some points of how it differese from other similar projects would be happy to include it. I think max three paragraphs would be about right. Open a PR on reply to this issue and I'll include it.

jmwright, 2020-11-03:
Thanks @Irev-Dev 

I think your summary is fine. There are two fairly common things that come up that set CadQuery apart:

* OpenCASCADE (OCCT) is a BRep kernel, which is a very powerful way to represent geometry. This is in contrast to OpenSCAD which uses the CSG kernel. Both approaches are great and people should use what they want, but I prefer working with BRep.
* Related to "close as possible to how you’d describe the object to a human" is the idea of "capturing design intent". For example, by using selectors instead of picking specific edges manually, the user can be sure the edges they intended to have fillets are always filleted, even if other aspects of the design changes.

I'm not sure if any of that is worth an edit to your list, but I thought I'd bring them up.

I can get you a more interesting screenshot than the block you have in the banner image, if you want. We have a 3D printer extruder mount and a CNC router spindle assembly that are interesting. It's totally up to you though.

Irev-Dev, 2020-11-05:
Thanks @jmwright ,
That sounds good I'll add some of that in. Can you point me to somewhere in the docs about selecting edges to fillet? I'd like to wrap my head around that a little more before I add it in.

adam-urbanczyk, 2020-11-05:
Take a look here regarding selectors
https://cadquery.readthedocs.io/en/latest/selectors.html#combining-selectors
and here regarding actual fillet usage:
https://cadquery.readthedocs.io/en/latest/examples.html#rounding-corners-with-fillet

Note that you can also select a face or a solid and execute a fillet op:
[image]

Irev-Dev, 2020-11-05:
Thanks @adam-urbanczyk,
Just what I was after.
So I've made some changes. The points about Brep I've added to the section on OpenCascade because really this applies to all of the projects that use it.

I also pulled an example from the docs @adam-urbanczyk  linked.

I'll merge this soon unless either of you wants further adjustments.
https://github.com/Irev-Dev/curated-code-cad/pull/2/files

jmwright, 2020-11-05:
@Irev-Dev Looks good to me. Thanks!

jmwright, 2020-11-09:
It looks like the changes have been merged into the list now, so I'm going to close this issue. @Irev-Dev feel free to reopen if needed, and thanks!

closed 2020-11-09 by jmwright