CadQuery/cadquery #495: extrude by rotating around axis ?

MarcWeber, 2020-11-03:
It's not in the planned features and I failed finding it ine the api.
PyMSH seems to implement it based on OCC: https://bthierry.pages.math.cnrs.fr/tutorial/gmsh/occ_spline/
I failed finding the OCC primitives, but probably I just don't know it well enough.

Can you add it to the examples/documentation or planned features ?

If OCC supports it can you point me to the name I missed to look it up ?

marcus7070, 2020-11-03:
Is `Workplane.revolve` what you are looking for?

https://cadquery.readthedocs.io/en/latest/classreference.html#cadquery.Workplane.revolve

jmwright, 2020-11-03:
Here's an example as well. https://github.com/CadQuery/cadquery/blob/master/examples/Ex022_Revolution.py

MarcWeber, 2020-11-03:
Yes, revolve is exactly what I was looking for. I missed it cause I only searched for extrude and alike.

https://cadquery.readthedocs.io/en/latest/apireference.html

Maybe the documentation text >>Use all un-revolved wires in the parent chain to create a solid.<< could be improved a little. But maybe this github issues is also enough to be found by people like me :-/

CadQuery is cool.

closed 2020-11-03 by MarcWeber