CadQuery/cadquery #500: Union result depends on order

bragostin, 2020-11-10:
I have followed the discussion on Google Groups and since it looks like a (kernel?) bug to me I am posting this issue here (I am using OCP 7.4).
In the following code, `CV = CV.union(lid)` and `CV = lid.union(CV)` yield different outputs, the later giving the flawed result shown in the screenshot.
This is a simplified example, but in reality I have a longer list of objects to union for which changing the order just makes the issue appear at a different location.
Without the symmetry cuts the issue disappears.
A workaround is using `union` with `clean=False`, but then gmsh is not able to mesh the resulting brep file.
```python
import cadquery as cq
from math import cos, sin, tan, sqrt, pi

scaleM = cq.Matrix([
[ 1., 0., 0., 0.],
[0., 1., 0., 0.],
[sin(20.*pi/180), 0., 1., 0.]])

sym0 = cq.Workplane('YZ').box(139., 65., 101.)
sym0 = sym0.translate((0., 0., 55./2.))
sym1 = sym0.translate((0., 139./2., -55./2))
sym0 = sym0.cut(sym1)
sym0 = sym0.val().transformGeometry(scaleM)
sym0 = cq.Workplane('XY').newObject([sym0])

CV = cq.Workplane('XY').rect(101., 139.).rect(100., 138.).extrude(55.)
CV = CV.val().transformGeometry(scaleM)
CV = cq.Workplane('XY').newObject([CV])
CV = CV.cut(sym0)

lid = cq.Workplane('XY').box(1., 139., 60.).translate((-100./2. ,0 , 60./2.*sin(20.*pi/180)))

CV = CV.cut(lid)

lid = lid.cut(sym0)

#CV = CV.union(lid)
CV = lid.union(CV)
```
[image]

jmwright, 2020-11-10:
It will be interesting to see if the [upgrade to OCCT 7.5](https://github.com/CadQuery/OCP/pull/30) has any impact on this.

adam-urbanczyk, 2020-11-11:
OCCT7.5 should be soon available - I just merged https://github.com/CadQuery/OCP/pull/30 . You could try if it helps @bragostin .

adam-urbanczyk, 2020-11-24:
I can confirm that this issue will be fixed by OCCT 7.5

[image]

bragostin, 2020-12-27:
I confirm that OCP 7.5 solves this issue and therefore close it. Bug fixing got a lot faster with pywrap!

closed 2020-12-27 by bragostin