CadQuery/cadquery #502: CadQuery consumes all system memory when trying to use `mirrorY`

TerranceN, 2020-11-11:
I just started learning CadQuery, and while building my first real part with it I ran into an issue where CadQuery all of a sudden freezes, eats up all my system's memory, and get's OOM-killed.

```
import cadquery as cq

pipeRadius = 23.5/2
thickness = 4
depth = 5

pipeClampOutline = (
    cq.Workplane("XZ")
    .circle(pipeRadius + thickness)
    .extrude(depth)
    .faces("<Y").workplane()
    .circle(pipeRadius)
    .cutThruAll()
    .faces(">Y").workplane()
    .moveTo(0, (pipeRadius+thickness)/2)
    .rect(2*(pipeRadius+thickness), pipeRadius+thickness)
    .cutThruAll()
)

pipeClampWithBlock = (
    pipeClampOutline.faces(">Z").faces(">X").workplane()
    .move(-thickness/2+5)
    .rect(10, 10)
    .mirrorY()
    .extrude(-5)
)

show_object(pipeClampWithBlock)
```

The problem started when I tried to add the `mirrorY`. Without that it runs fine. Also I might be using `mirrorY` incorrectly (was just trying to learn how it works), but the program responding like this still seems like a bug.

I'm using CQ-editor, but I also tried running the python file directly, which I think bypasses the editor (meaning it would be an issue with CadQuery itself, not CQ-editor, which is why I'm opening this issue here).

I have the latest CQ-editor, which looks like it just pulls the latest CadQuery, so I should be on the latest version of that too, and I'm running Ubuntu 19.10.

TerranceN, 2020-11-11:
As a workaround I only generated one half of my design and then mirrored that side once it was done rather than trying to mirror each operation. I've gotten it to work, but not before running into another crash. Mirroring around the origin so that both halves line up perfectly froze CadQuery in some kind of infinite loop (stuck with high cpu usage), but memory usage was stable.

```
pipeRadius = 23.5/2
thickness = 4
depth = 5

m3ThroughHoleRad = 3.25/2

pipeClampOutline = (
    cq.Workplane("XZ")
    .circle(pipeRadius + thickness)
    .extrude(depth)
    .faces("<Y").workplane()
    .circle(pipeRadius)
    .cutThruAll()
    .faces(">Y").workplane()
    .moveTo(0, (pipeRadius+thickness)/2)
    .rect(2*(pipeRadius+thickness), pipeRadius+thickness)
    .cutThruAll()
    .moveTo(-(pipeRadius+thickness)/2, 0)
    .rect(pipeRadius+thickness, 2*(pipeRadius+thickness))
    .cutThruAll()
)

pipeClampWithBlock = (
    pipeClampOutline.faces(">Z").faces(">X").workplane()
    .move(thickness/2-5).workplane()
    .rect(10, 10)
    .extrude(-5, combine=False)
    .faces("<Z").workplane().tag("endOfExtrusion")
    .union(pipeClampOutline)
    .workplaneFromTagged("endOfExtrusion")
    .circle(m3ThroughHoleRad)
    .cutThruAll()
    .workplaneFromTagged("endOfExtrusion")
    .polygon(6, 5)
    .cutBlind(-2)
    .workplaneFromTagged("endOfExtrusion")
    .polygon(6, 5)
    .cutBlind(2)
)

pipeClampWithBlockMirrored = pipeClampWithBlock.mirror(mirrorPlane="YZ", basePointVector=(0, 0, 0))

pipeClamp = pipeClampWithBlock.union(pipeClampWithBlockMirrored)

show_object(pipeClamp)
```

When I change the basePointVector such that the two halves are overlapping a bit (`basePointVector=(-0.01, 0, 0)`), it works fine.

bragostin, 2020-11-12:
I think it's because the `rect` your are mirroring yields 2 overlapping rectangles.
If you use `move(-thickness/2+5+2)`, the 2 rectangles are not overlapping and the memory issue disappears.

adam-urbanczyk, 2020-11-12:
Thanks for reporting @TerranceN , these are two separate CAD kernel issue. The first one occurs in the clean step, I think that you put the rectangles at an unintended location (it should not cause a crash/out of memory of course). I did not check the second take in detail. To get you unblocked, here is how I'd model the part (as far as I can understand your design intent):
```python

import cadquery as cq

pipeRadius = 23.5/2
thickness = 4
depth = 5
m3ThroughHoleRad = 3.25/2
r = 10

pts = [(-pipeRadius-thickness-1,0),(pipeRadius+thickness+1,0)]

pipeClampOutline = (
    cq.Workplane("XY")
    .moveTo(pipeRadius+thickness/2)
    .rect(thickness,depth)
    .revolve(180,(0,0,0),(0,-1,0))
)

pipeClampWithBlock = (
    pipeClampOutline.faces("<Z").workplane()
    .pushPoints(pts)
    .rect(r,r)
    .extrude(-5)
    .faces('<Z[-2]')
    .workplane()
    .pushPoints(pts)
    .circle(m3ThroughHoleRad).cutThruAll()
    .faces('<Z[-2]')
    .workplane(offset=-3)
    .pushPoints(pts)
    .polygon(6, 5)
    .cutBlind(pipeRadius)
)
```
[image]

TerranceN, 2020-11-12:
Wow, thank you for taking the time to re-write that! There's lots for me to learn from even that small snippet!

And yeah, it I was thinking `mirrorY` would mirror globally, but now I see it does it based on the current workplane (which seems more sensible).

lorenzncode, 2022-09-23:
I cannot reproduce the problem with the first code example testing with current master (ocp-7.5.3).

[image]

adam-urbanczyk, 2022-09-23:
Thanks @lorenzncode , closing.

closed 2022-09-23 by adam-urbanczyk