CadQuery/cadquery #505: How to avoid spline to start from origin in cadquery script

rasunag27, 2020-11-13:
Dear all,

I am working on some tubular structures in which I am using splines and sweep to create the path of points.

But, the points does not start from origin but from different (x, y) coordinates. When I give spline to those points, the spline does well but it creates a closure by reaching to origin. How do I avoid that.?

Example:

In the below example, I want the spline to start from (1, 2), but it start from (0,0). How to remove the spline from origin?

`import cadquery as cq
from Helpers import show
Points we will use to create spline and polyline paths to sweep over

pts = [(1, 2), (2, 1), (4, -1)]
Spline path for points

path = cq.Workplane("XZ").spline(pts)

show(path)`

I have attached the image link for reference.

[Spline](https://drive.google.com/file/d/1aLy0HonZ6YMIcyhoG-ZMyQ7aUzbJMaS2/view?usp=sharing)

Any leads will be appreciated.

Thanks and Regards,

Sunag R A.

jmwright, 2020-11-13:
You'll want to use `moveTo()` in your code. Something like this:
```
import cadquery as cq

pts = [(2, 1), (4, -1)]

path = cq.Workplane("XZ").moveTo(1, 2).spline(pts)

show_object(path)
```
An example of using a spline to define the edge of a plate is [here](https://github.com/CadQuery/cadquery/blob/master/examples/Ex010_Defining_an_Edge_with_a_Spline.py)

I would also recommend switching from the FreeCAD module to [CQ-editor](https://github.com/CadQuery/CQ-editor#installation) as your GUI, if possible.

rasunag27, 2020-11-13:
Thank you very much for the quick response. !!

I will definitely try with CQ-editor.

Regards,
Sunag R A.

adam-urbanczyk, 2020-11-13:
You can also use `includeCurrent=False` in your spline call.