CadQuery/cadquery #506: How to combine all CQ Models and export to stl file?

rasunag27, 2020-11-16:
Dear All,

I have a tubular tree like structure with branches obtained from my python script. I am new to this cadquery so I have written in the free format rather than functions in python as of now. 

I want to combine all the objects obtained from the script and export/save as an stl file. All I could use in combineSolids and union. But by using this, the code is eliminating some elements due to which I am not getting how to combine as a single object. 

I have attached the code link here:

[Code link](https://drive.google.com/file/d/1nspdZIdLnXfwCZn21o9tKItsROPm0V9_/view?usp=sharing)

Here the object contains sweep, sweep2, 2a, 2b, 2c, sweep3, 3a, 3b, 3c and sweep4,4a and 4b. I need to combine all these objects as a single object. I could object some files combined using union and combineSolids, but not all. 

Also I have attached the image which I need to obtain as a single file.

[Image](https://drive.google.com/file/d/19UfkS-UzLIe2M9AGq8e9QbbGxPNToTaR/view?usp=sharing)

Any leads will be appreciated.

Regards,
Sunag R A.

jmwright, 2020-11-16:
If you want them to just be part of a single Workplane object, you could do this.
```
all_children = cq.Workplane().newObject([sweep2.val(), sweep2a.val(), sweep2b.val(), sweep2c.val(), sweep3.val(), sweep3a.val(), sweep3b.val(), sweep3c.val(), sweep4.val(), sweep4a.val(), sweep4b.val()])
```

rasunag27, 2020-11-17:
Dear Jeremy,

Thank you so much for the reply. I implemented this and I found one small Type Error in __init__ method. I tried exploring the source code for the same but could not find one. Could you walk me through this?

Error:

Type Error: __init__() missing 1 required positional argument: 'inPlane'

Regards,

Sunag R A.

rasunag27, 2020-11-17:
Dear Jeremy,

Thank you very much for the reply. I implemented this and it is working fine.

But I have one problem after saving the file to stl format. Below is the code and the stl file viewed in Paraview after saving.

`all_children = cq.Workplane("XY").newObject([sweep2.val(), sweep2a.val(), sweep2b.val(), sweep2c.val(), sweep3.val(), sweep3a.val(), sweep3b.val(), sweep3c.val(), sweep4.val(), sweep4a.val(), sweep4b.val()])

#### Export to stl #####
all_children.val().exportStl("all_children.stl")`

The export did not give me any problem. But, when I load the stl in Paraview for viewing it did not show the entire model. Here is the image link.

[stl_in_paraview](https://drive.google.com/file/d/1oZbthUxzLDCi08ngei9vP-P1jdUp9NMf/view?usp=sharing)

Please walk me through this.

Regards,
Sunag R A.

jmwright, 2020-11-17:
There seems to be something wrong with STL export in this case. I exported to both STL and STEP using the following form, which I would suggest.
```
from cadquery import exporters
exporters.export(cmpd, "/home/jwright/Downloads/vascular.step", exporters.ExportTypes.STEP)
```
When I open the exported STL in FreeCAD I see this:
[image]

But the STEP looks correct:
[image]

I wonder if @adam-urbanczyk has any ideas why the STL export isn't working correctly.

rasunag27, 2020-11-18:
Dear Jeremy,

Thanks for trying it out and obtaining the same output.

I will check with other means. Meanwhile, @adam-urbanczyk let us know your ideas / other ways related to this.

Thanks and Regards,
Sunag R A.

adam-urbanczyk, 2020-11-18:
This worked for me:
```
result = cq.Workplane().add(child1).add(child2).add(child3).add(sweep3c)
cq.exporters.export(result,'vein.stl',tolerance=2e-3, angularTolerance=1e-2)
```
[image]

jmwright, 2020-11-22:
I think there are a couple of defects in that STL if you look closely, but I'm guessing playing with the `tolerance` and `angularTolerance` may fix it.

jmwright, 2021-02-26:
This issue is a question and has been dormant since Nov 2020, so I'm closing it.

closed 2021-02-26 by jmwright