CadQuery/cadquery #507: How to sweep multisection along 3D..?

rasunag27, 2020-11-16:
Dear all,

I would like to sweep from one diameter to another in a 3D space using points and spline. I tried using "Sweep_with_multisection" example. But, this uses offset in a single coordinate which is not useful for my case.

Below, I have mentioned the code in which the base I need diameter of 4mm and top I need diameter of 2 mm. 

`import cadquery as cq
from cadquery import exporters
show = show_object

#circle = cq.Workplane("XY").circle(0.072)
#show(circle)

dia = 2
x = 50
y = 0
z = -5

pts = [(x, y, z), (x, (y+2), z), ((x-2), (y+14), z)]
path= cq.Workplane("XY").moveTo(x, 0).spline(pts)
sweep = cq.Workplane("XZ").moveTo(x, z).circle(dia).sweep(path)`

From the above code, I get the entire tube as diameter 2. Below is the manipulated code for sweep.

`sweep =  cq.Workplane("XZ").moveTo(x, z).circle(4).workplane(offset = (x-2, y+14, z)).circle(dia).sweep(path, multisection=True)`.

How to modify this sweep code so as to obtain base dia as 4mm and top dia as 2mm.

Below is the reference image.

[Sweep_link](https://drive.google.com/file/d/1jno-Yb3Rq0xpfBWTqriQ9zAtHltoAgov/view?usp=sharing)

Any leads will be appreciated.

Regards,
Sunag R A.

jmwright, 2020-11-16:
There may be a cleaner way to do it, but have you looked at using loft? https://cadquery.readthedocs.io/en/latest/examples.html?highlight=loft#making-lofts

With lofts I expect you'd need to break the sweep up into sections. I'm not sure if that's useful either for your case.

adam-urbanczyk, 2020-11-17:
You might want to try this:
```python
dia = 2
x = 50
y = 0
z = -5

pts = [(x, y, z), (x, (y+2), z), ((x-2), (y+14), z)]
path= cq.Workplane("XY").moveTo(x, 0).spline(pts)

sweep = (cq.Workplane("XY")
    .pushPoints([path.val().locationAt(0)]).circle(2)
    .pushPoints([path.val().locationAt(1)]).circle(4)
    .consolidateWires()
    .sweep(path,multisection=True)
    )
```
[image]

rasunag27, 2020-11-18:
Dear adam,

Thank you for the reply.

I tried this and I am facing an attribute error stating 'Edge' object has no attribute named 'locationAt'. 

What needs to be imported so as to remove the error.??

Thanks and Regards,

Sunag R A.

marcus7070, 2020-11-18:
`locationAt` is a very recent addition to CadQuery. If you are working from the stable version, you will have to switch to the master version as described in the `README.md`.

rasunag27, 2020-11-18:
Dear @marcus7070 ,

Thank you very much for the reply. I have installed the Cadquery using master only, Here is the same installation code line I have used as said in `README.md` .

`conda install -c conda-forge -c cadquery cadquery=master`

Or maybe do I need to reinstall again?

Regards,

Sunag R A.

adam-urbanczyk, 2020-11-18:
Try reinstalling:
```
conda clean --all --force-pkgs-dirs
conda install --force-reinstall -c conda-forge -c cadquery cadquery=master
```

This sadly so complicated due to this conda issue: https://github.com/conda/conda/issues/7741

rasunag27, 2020-11-26:
@adam-urbanczyk  I tried re-installing cadquery. But I am not able to get the attribute `locationAt`.

Are there any other alternative ways to obtain?

Regards,
Sunag R A.

adam-urbanczyk, 2020-11-26:
Could you paste the output of the above commands?

As a last resort you can try:
```
pip install git+https://github.com/CadQuery/cadquery.git
```

rasunag27, 2020-11-27:
@adam-urbanczyk Thank you for the git link.

I did install the git and the cadquery is working now. After this installation, I think "locationAt" is working. But the display of the CAD model is giving me an error. Meaning, when I run the script it does not display anything throwing and error in the Log Viewer. 

I researched and could not been able to address the error. The error is presented below.

`ERROR: CQ GUI: Uncaught exception occurred
Traceback (most recent call last):
  File "/home/sharat/anaconda/lib/python3.6/site-packages/cq_editor/widgets/object_tree.py", line 260, in addObjects
    ais,shape_display = make_AIS(shape)
  File "/home/sharat/anaconda/lib/python3.6/site-packages/cq_editor/cq_utils.py", line 35, in make_AIS
    ais = AIS_ColoredShape(shape.wrapped)
  File "/home/sharat/anaconda/lib/python3.6/site-packages/OCC/Core/AIS.py", line 15019, in __init__
    _AIS.AIS_ColoredShape_swiginit(self, _AIS.new_AIS_ColoredShape(*args))
TypeError: in method 'new_AIS_ColoredShape', argument 1 of type 'Handle_AIS_Shape const &' `

Please let me know if any leads are present related to this. 

Regards,

Sunag R A.

rasunag27, 2020-11-27:
Dear @adam-urbanczyk ,

After lot of uninstall, reinstall and creating new env for conda, the current version is working with sweep method.

Thank you for the support.

Regards, 
Sunag R A.